UniversalRobots/Universal_Robots_ROS_Driver #206: robot_driver: unused / non-existing 'vel_based_pos_traj_controller'?

gavanderhoorn, 2020-06-23:
The `ur_control.launch` file lists a controller called `vel_based_pos_traj_controller` in the `default` of its `controllers` arg:

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/bace1d6ed881a688b4353d627160a78668356584/ur_robot_driver/launch/ur_control.launch#L16

This doesn't appear to exist (or isn't listed as one of the controllers in the `ur_robot_driver/config` directory).

It's also mentioned in the documentation:

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/ae865a94067316e0ef4c3b8fbfb26f383fca0375/ur_robot_driver/doc/ROS_INTERFACE.md#L161

There is a `vel_traj_controller` ([here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/bace1d6ed881a688b4353d627160a78668356584/ur_robot_driver/config/ur10_controllers.yaml#L101-L123)) though, but that's only ever defined in the `.yaml` configuration files, never loaded nor spawned by the `.launch` files.

fmauch, 2020-06-23:
It was by intention that it is not in the default launch files as it was never tested thoroughly (see #119 and #124), so I didn't want to add them to any default set (neither the started nor the stopped controllers), so people only activated it if they really wanted to.

Looks like I missed to bring the default values in `ur_control.launch` in line with that decision.

blackhorsewu, 2020-09-16:
Dear @gavanderhoorn and @fmauch, I have just tried to use the `vel_traj_controller`. Is it working properly according to your initial intend? I expected it works like the `vel_based_controller` back in the `ur_modern_driver`. I have just posted a question in the answers.ros.org.

gavanderhoorn, 2020-09-16:
> I have just tried to use the `vel_traj_controller`. Is it working properly according to your initial intend? I expected it works like the `vel_based_controller` back in the `ur_modern_driver`. 

Could you be a bit more specific?

What is the `vel_based_controller`? Do you mean the `vel_based_pos_traj_controller` (ie: `velocity_controllers/JointTrajectoryController`) or the `joint_group_vel_controller` (ie: `velocity_controllers/JointGroupVelocityController`).

The first is a joint trajectory controller which outputs velocities to reach a position, while the second is a forwarding-controller which takes in velocity setpoints and simply forwards it to the robot controller.

Use of these would be rather different.

blackhorsewu, 2020-09-16:
@gavanderhoorn , thank you. I am writing the new issue now.

closed 2021-08-03 by fmauch