UniversalRobots/Universal_Robots_ROS_Driver #213: Consider renaming '*_traj_controller' to *_joint_traj_controller'

gavanderhoorn, 2020-06-30:
(I've removed the sections suggested by the template to avoid making this an overly long issue)

As per subject.

Other controllers in [ur_robot_driver/config](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/06feddde1081dfda65d3a138f6bca572f4f4e8de/ur_robot_driver/config) files have names which reflect their types, such as `joint_state_controller`, `speed_scaling_state_controller` and `joint_group_vel_controller`.

The `pos_traj_controller`, `scaled_pos_traj_controller`, `vel_traj_controller` and `scaled_vel_traj_controller` however leave out the `joint` part.

To be more explicit about *which* type of controller is configured by the stanzas following that name, and to make it easier to figure out (or *remember*) which controller type is being used when working with the `controller_manager` command line/UI tools and when configuring MoveIt or other clients of these controllers it would be good for those names to include the `joint` part.

Suggestion: change the names to the following:

 - `pos_joint_traj_controller`
 - `scaled_pos_joint_traj_controller`
 - `vel_joint_traj_controller`
 - `scaled_vel_joint_traj_controller`

gavanderhoorn, 2020-06-30:
Another reason for doing this would be [fzi_robot_interface_proposal](https://github.com/fzi-forschungszentrum-informatik/fzi_robot_interface_proposal): if/when a controller (or multiple) will be created using that interface, they will likely be added to the default set here in this repository. `pos_cart_traj_controller` seems like it would be a good name to distinguish it from the `pos_joint_traj_controller`.

gavanderhoorn, 2020-06-30:
Some context: this occurred to me while working on ros-industrial/universal_robot#518.

fmauch, 2020-06-30:
yes, that's a valid suggestion.

gavanderhoorn, 2020-08-05:
#214 got merged, so this can be closed.

closed 2020-08-05 by gavanderhoorn