UniversalRobots/Universal_Robots_ROS_Driver #215: Switching between Physical Robot and URSim lead to unexpected movement

rolling2020, 2020-07-01:
# Summary
It appears that switching from using a physical robot to a URSim simulated robot on the same system without stopping the teach pendant program can lead to unexpected movement on the physical robot.

I used the driver with a physical UR5. I wanted to allow my colleague to use the physical robot, so I attempted to switch to using a URSim robot: 
I closed the driver that was connected to the physical robot; this included closing the ROS master. However, I did not stop the External Control program that was being executed on the teach pendant, as I did not think there was a need to so at this point. 
I then started ursim.
I then used my launch file for starting the driver to use the localhost IP rather than the IP of the physical robot, which started a new ROS master with it.

It was at this point that the unexpected movement occurred:  the physical robot moved to match the position of the URSim robot, despite the ROS driver that had been connected to the physical robot already being closed. 

It's unclear to me, but I am assuming some kind of connection was left over on the port that the new driver then took over?

I am assuming a workaround to avoid this would be to simply stop the execution of the program on the teach pendant, or perhaps to use different "reverse_port"/"custom port"s for the simulated robot and physical robot, but confirmation on that would be greatly appreciated. I was lucky that there was no collision with the environment this time, but I'd like to make sure I don't experience this again in the future.

# Versions
 - ROS Driver version: ur_robot_driver package version 0.0.3 (git commit a926789adc0cefc616beb0fda15793cdcd65831b)
 - Affected Robot Software Version(s): 3.13.1.10297 (physical UR5 & URSim)
 - Affected Robot Hardware Version(s): UR5 CB3 + URSim
 - URCaps Software version(s): 1.0.1

fmauch, 2020-07-01:
The program on the TP is ended as soon as the driver is shutdown on the ROS side.

However, if you have the "Program loops forever" option enabled on the TP (Which I think is default), then it will try to reconnect in an endless loop.

While this isn't optimal, this should only request the URScript from the driver, but not receive any motion commands, as these are sent to the explicit socket opened to the IP address given to the driver.

I'll try to reproduce that, as it definitely seems suboptimal.

fmauch, 2020-07-01:
I cannot reproduce that. Doing the steps you described yields to the expected output on the driver repeating 

```
[ INFO] [1593593369.811958728]: Robot ready to receive control commands.
[ INFO] [1593593371.813841932]: Connection to robot dropped, waiting for new connection.
[ INFO] [1593593371.817923895]: Robot ready to receive control commands.
[ INFO] [1593593373.819835009]: Connection to robot dropped, waiting for new connection.
```

as the robot requests the program (through the script_sender_port), gets it, sends a keepalive signal to the driver (through the reverse_port) (which is why it is reporting "Robot ready to receive control commands.") and then drops the connection, as no command is received in the socket timeout.

As having said earlier: This behavior isn't optimal, but that's what I would expect from the way the driver should be working.

Note: I tested this with an e-Series robot, but I don't see a reason this should be different with a CB3 version.

However, the driver version you are using is quite old, you might want to consider updating and test this again.

stefanscherzinger, 2021-08-09:
@rolling2020 Is this still relevant?

rolling2020, 2021-08-10:
I can confirm that I have not experienced this issue since updating the
driver version last year. It appears that a process for ending the
connection (or perhaps it is verifying the current connection?) with the
Teach Pendant software was added at some point after the old version I
mentioned in my original post, and that likely fixed the issue.

With that version I had been using, if I used the ROS driver to control the
robot and then ended the driver program but left the URCaps program running
on the Teach Pendant, then restarted the ROS driver program, it was able to
control the robot without me ever touching the URCaps program.
With more recent versions, if I use the ROS driver to control the robot and
then end the driver program but leave the URCaps program running on the
Teach Pendant, then restart the ROS driver program, I have to restart the
URCaps program before I am able to control the robot again. While it is a
tiny bit more of a hassle to use, I think this is likely what is preventing
the issue I mentioned in my original post, so it is a welcome change.

@fmauch Thank you for your help on this issue.

closed 2021-08-10 by stefanscherzinger