UniversalRobots/Universal_Robots_ROS_Driver #216: Can't control Ur3 with rqt_joint_trajectory_controller

EiGa01, 2020-07-01:
# Summary

# Versions
 - ROS Driver version: 
 - Affected Robot Software Version(s): 
 - Affected Robot Hardware Version(s): UR 3
 - Robot Serial Number: 2017332081
 - UR+ product(s) installed:
 - URCaps Software version(s): 3.12.0.90886

# Impact
I followed the instructions of the ReadMe and was not able to connect the tower to the robot: see the following link: 
[Problem posted in ROS Answer](https://answers.ros.org/question/356106/cant-control-ur3-out-of-rviz/)

gavanderhoorn, 2020-07-01:
> Everything works well with exception of the real time , but I want to try it in general, before get involved with realtime, and in the instruction is written, that it in general should work anyway.

@EiGa01: could you please post the complete output you receive after `roslaunch`-ing the main driver `.launch` file?

Put it in a code-block ([docs](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks)) to keep the formatting correct.

EiGa01, 2020-07-02:
@gavanderhoorn: I edited in the first post

fmauch, 2020-07-02:
This looks like you are missing to start the corresponding program on the robot. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#i-started-everything-but-i-cannot-control-the-robot

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md explains how to create a program containing an "ExternalControl" program node. Start that program once the driver is started.

EiGa01, 2020-07-02:
Well, I tried this new installation three times, and nothing changed...

Now I pulled the new version up externalcontrol-1.0.2.urcap and everything works. 
I'm not sure if I was too stupid three times and it needed a fourth try, but now I'm able to connect with the rqt_joint_trajectory_controller

Thank you very much :)

closed 2020-07-02 by EiGa01