UniversalRobots/Universal_Robots_ROS_Driver #218: Reproducing simulated poses in ROS

akosv95, 2020-07-04:
Hello

I am trying to implement a task-based kinematic controller in Python. I am using ROS Melodic on Ubuntu 18.04, and simulating an UR10 with URSim 5.8.2. I want to be able to calculate the positions and velocities of different points of the robot based on the current joint states. 

Currently I am trying to do this with the pykdl_utils package: http://wiki.ros.org/pykdl_utils
I can access the joint states of the simulated robot through the /joint_states ROS topic, and the URDF description of the robot, as in the examples. However, calculating the position of the end effector using these does not match the values shown in URSim. Originally I got the wrong values on everything, but after changing the base_correction value in the ur10 URDF file in ur_description from pi to 0, I get correct values on X and Y coordinates for the end effector, but the Z value is still off for some reason.

I have also tried to do these things with pinocchio instead of pykdl, but didn't have any more success. I suspect that there is some discrepancy between the URDF descriptions and the simulated robot, but don't know how to troubleshoot further...
(The joint positions match perfectly in ROS and URsim, and I am passing them in the correct order for the kinematics calculation).

gavanderhoorn, 2020-07-04:
It might help if you could show some of the results you get which you believe are incorrect.

Also: *only* the `base->tool0` transform will correspond to what the TP shows, and then only if you have the TP configured to show you the pose in the *Base` "feature" (== frame) and have an all-zeros TCP configured.

`base->tool0_controller` should give you the transform the TP shows (again, when you have it configured to show it in the *Base* feature) and only if the driver is active.

akosv95, 2020-07-04:
Right, the problem was non-base feature in the simulator. My bad. Thank you very much for the help!

closed 2020-07-04 by akosv95