UniversalRobots/Universal_Robots_ROS_Driver #219: Force/Impedance control of UR e-series

deepakraina99, 2020-07-06:
Is it possible to do cartesian impedance control using ROS for UR e-series? I am able to do with real robot using teach pendant but I wanted it to do through ROS control?
If not, do I have force feedback in UR series (in ROS topics) which can be used to write my own impedance controller?

Thanks.

fmauch, 2020-07-06:
Currently, this is not supported, no.

You get the wrench published via the `force_torque_sensor_controller` which is started by default and publishes the wrench on `/wrench` (also see #218).

As there is no Cartesian control in ROS-Control (yet), you'll either have to do the inverse kinematics yourself or use something like https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers to get it working.

As we are currently working on a Cartesian controller, this might get things s bit easier to implement soonish (around the end of this year, I assume).

deepakraina99, 2021-01-07:
> Currently, this is not supported, no.
>
> You get the wrench published via the `force_torque_sensor_controller` which is started by default and publishes the wrench on `/wrench` (also see #218).
>
> As [...]

@fmauch Would you be able to implement cartesian control ?

fmauch, 2021-01-09:
Colleages of me are working on it right now.

closed 2021-01-09 by deepakraina99

deepakraina99, 2021-09-04:
@fmauch  Any updates on this ?

urrsk, 2021-09-09:
@deepakraina99 Currently, It is not possible to program your own impedance control loop as the UR controller only supports position and velocity commands and not torque commands. But I am aware of several successful implementation of admittance control loop using both `servoj` and `speedj`.

fmauch, 2021-09-09:
BTW, Cartesian control is now included into this driver.

ZZWang21, 2022-08-25:
So, any update? Or anyone has achieved this? I have received the force signal and can adjust the position of the end-effector. How should I combine them? Thank you.

2000222, 2022-11-08:
> 

Hope new update....But could u tell us some hint or tutorials about how to receive the force signal and adjust the position of the end-effector~~ Thank u

stefanscherzinger, 2022-11-08:
@2000222 @deepakraina99 

In case you do not want to write your own controller, let me point to this [control suite](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers#cartesian-controllers). It works great for end-effector force control (among other use cases) on the UR and URe series. [Here](https://github.com/stefanscherzinger/cartesian_controllers_universal_robots)'s an example package on a UR10e.

2000222, 2022-11-09:
> @2000222 @deepakraina99
>
> In case you do not want to write your own controller, let me point to this [control [...]

Thanks a lot! This control suite is so amazing 🤗🤗 and helpful🍻🍻😻😻

urfeex, 2025-11-14:
The link in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/219#issuecomment-1307308962 is probably still one of the most helpful.

AdrianPrados, 2025-11-14:
Thanks for the quick response, my question is if I need to run that controller suit in my computer and connect with my robot by ROS or I need to run it directly on my robot. Thanks for your help.