UniversalRobots/Universal_Robots_ROS_Driver #220: joint_limits.yaml file and saftey configuration on ur5 control panel

ShuSewts, 2020-07-07:
I've been asked to post here, here's the original [question:](https://answers.ros.org/question/356557/joint_limitsyaml-file-and-saftey-configuration-on-ur5-control-panel-am/)

I am using ROS Kinetic, moveit, and a ur5 with a polyscope version of 3.1.0.0.76181. 
I have changed the safefty configuration on the ur5 control panel to limit the maximum joint speed to 80 degrees/second. This resulted in some stuttering when moving the robot with moveit and a very quick acceleration (the robot started to move almost immediately at a fast speed and decelerated very quickly near the end of its trajectory) so I changed the default values in the joint_limits.yaml file inside the ur5_moveit_config to match the joint speed limits on the control panel. As no acceleration values can be found on the control panel, I changed the acceleration on the joints to be the same as the velocity. The robot is now able to accelerate smoothly, but I am still getting some stuttering in trajectories. Can anyone point me in the right direction? Attached my joint_limits.yaml file for reference. 

`# joint_limits.yaml allows the dynamics properties specified in the URDF to be overwritten or augmented as needed
# Specific joint properties can be changed with the keys [max_position, min_position, max_velocity, max_acceleration]
# Joint limits can be turned off with [has_velocity_limits, has_acceleration_limits]
joint_limits:
  shoulder_pan_joint:
    has_velocity_limits: true
    max_velocity: 1.2211
    has_acceleration_limits: true
    max_acceleration: 1.2211
  shoulder_lift_joint:
    has_velocity_limits: true
    max_velocity: 1.2211
    has_acceleration_limits: true
    max_acceleration: 1.2211
  elbow_joint:
    has_velocity_limits: false
    max_velocity: 1.2211
    has_acceleration_limits: true
    max_acceleration: 1.2211
  wrist_1_joint:
    has_velocity_limits: true
    max_velocity: 1.2211
    has_acceleration_limits: true
    max_acceleration: 1.2211
  wrist_2_joint:
    has_velocity_limits: true
    max_velocity: 1.2211
    has_acceleration_limits: true
    max_acceleration: 1.2211
  wrist_3_joint:
    has_velocity_limits: true
    max_velocity: 1.2211
    has_acceleration_limits: true
    max_acceleration: 1.2211`

gavanderhoorn, 2020-07-07:
Could you please use a [code block](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) to format the `.yaml` file?

fmauch, 2020-07-09:
Which controller are you using? The scaled version or the non-scaled `pos_joint_traj_controller`?

And just to confirm, I assume, you are running Polyscope version 3.10.0.76181, not 3.1.0.0.76181?

ShuSewts, 2020-07-09:
I am using the scaled version of `pos_joint_traj_controller` and yes the polyscope version is 3.1.0.0.76181.

fmauch, 2020-07-10:
> I am using the scaled version of `pos_joint_traj_controller` and yes the polyscope version is 3.1.0.0.76181.

Both parts of your answer confuse me.

What is the name of the active controller?
 * `pos_joint_traj_controller`
 * `scaled_pos_joint_traj_controller`

Polyscope versions usually consost of four numbers, yours has 5. This is why I assumed, that you were using version 3.10.0.76181.

ShuSewts, 2020-07-24:
Sorry for the late reply and confusing answer, name of the active controller is `scaled_pos_joint_traj_controller` and I am using polyscope version 3.10.0.76181.

stefanscherzinger, 2021-09-01:
@ShuSewts Sorry, that issue got forgotten. Kinetic is no longer officially supported. Did you experience this issue with the latest driver version?

closed 2023-04-11 by github-actions[bot]