UniversalRobots/Universal_Robots_ROS_Driver #224: Linear Trajectories

saalimperator, 2020-07-08:
# Summary

Hello developers,
currently I am programming a ROS framework, or rather porting an existing one to the UR5. 
I have already migrated the teaching and PTP movements and am currently having a hard time finding a suitable controller for the linear movements. In Issue #121 a future realization has been teased. What is the current status there, as I cannot continue without this feature or has the problem been solved otherwise?
Best regards

fmauch, 2020-07-08:
The current status is still in the interface definition [here](https://fzi-robot-interface-proposal.readthedocs.io/en/latest/)

On the implementation side we started on the ROS-Control interface but that's not yet in a usable state. Time frame for this is end of the year.

saalimperator, 2020-07-13:
Thank you very much for the answer and the further information about the linear controller. I am looking forward to testing it in my framework. 
Best regards

closed 2020-07-13 by saalimperator

tonkei0361, 2020-07-29:
@saalimperator Can I ask how you implemented PTP motion with UR robots? Did you use another controller or make your own controller? Any help is appreciated! FYI, my robot is UR5e.

saalimperator, 2020-07-29:
@tonkei0361 Have you seen the chapter 'quick start' at README.md? -> https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#quick-start
If you need further information, you can look at -> http://wiki.ros.org/actionlib_tutorials/Tutorials/Writing%20a%20Simple%20Action%20Client%20%28Python%29
Also there is an Code example -> https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/testing_scripts

tonkei0361, 2020-07-29:
@saalimperator Thank you for your response! I am aware of the controller `scaled_pos_joint_traj_controller`, which sets individual joint angles of the robot. What I need for my application is to move the TCP position in Cartesian coordinate. When you said PTP motion, I thought that you were able to achieve this function. My current understanding is that I will need to calculate forward kinematics of the robot to obtain the TCP's xyz position. 

Sorry if my previous question wasn't clear enough and thank you again for your time to answer my questions!