UniversalRobots/Universal_Robots_ROS_Driver #226: Using IK Solvers Other Than KDL With Calibration Data

jbeck28, 2020-07-14:
I have been trying to find a closed-form kinematics solver for a UR5e that I could use with Descartes. I have managed to compile an IKFast solver and it seems to work perfectly in simulation. It isn't entirely clear to me where the calibration information gathered from the controller gets applied. It does not seem to get merged into the URDF, so I'm curious if the IKFast solver I made against a default URDF will still give me accurate positions when using this driver? 

I'd also like to thank the developers and maintainers for this project, it is really fantastic.

cpt-yoshi, 2020-07-15:
> It isn't entirely clear to me where the calibration information gathered from the controller gets applied.

See this [section](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/master/ur_calibration) to apply the calibration correction.

> It does not seem to get merged into the URDF...

Yes it does, but it requires that you add your `calibration.yaml` to the `kinematics_config` parameter.

> So I'm curious if the IKFast solver I made against a default URDF will still give me accurate positions when using this driver?

When using the **default** URDF, the IK will not give you accurate positions for your real robot. The calibration is not added by default.

I have tried to generate an IK with IKFast with my calibration but it failed. I suspect that it is related to this: a 6 DOF robot needs the first 3 joints or the last 3 joints axes intersecting at a common point ([link](https://canyonkang.wordpress.com/portfolio/robotics-inverse-kinematics-implementation-kdl-ik-vs-ik-fast/)). I have not investigated this further...

As a side note you should try the [track-ik](https://bitbucket.org/traclabs/trac_ik/src/master/) solver. You should see a performance boost in speed and solving rate.

jbeck28, 2020-07-17:
Thank you for your reply and information. 

To be clear, my confusion was that, to my knowledge of IKFast (limited), one must generate an IK solver against a particular URDF. In this application, it wasn't clear to me where in the URDF to input the kinematics adjustments contained in the mentioned yaml file. I had thought that the URDF which gets placed on the parameter server at `/robot_description` would have already been modified with the changes contained in that `calibration.yaml` file generated from information stored on the robot controller. This did not seem to be the case to me based on what I saw. So it is still a bit unclear to me where this calibration information gets used.

You mention that you have tried to generate an IKFast solver which took into account the calibration information; I would be very thankful if you could point me in the direction of how you included this calibration information in your URDF!

I did manage to create an IKFast solver for my UR5e without the kinematic calibration information, and it works well enough for my application (and enabled me to use Descartes, as desired). However, I did not do anything to verify how accurate it was in returning the actual position of the robot. I suppose a good test would be to perform IK on a given pose of the actual robot arm, then perform fk on the returned joint coordinates and compare the values? 

Thanks again for the help!

cpt-yoshi, 2020-07-17:
> You mention that you have tried to generate an IKFast solver which took into account the calibration information; I would be very thankful if you could point me in the direction of how you included this calibration information in your URDF!

Quick way to generate a URDF that includes the calibration of your robot: 
``` terminal
rosrun xacro xacro -o calibrated.urdf ur5_robot.urdf.xacro kinematics_config:=/<YOUR_PATH>/calibration.yaml
```

> I suppose a good test would be to perform IK on a given pose of the actual robot arm, then perform fk on the returned joint coordinates and compare the values?

Yes or you can compare the desired tcp pose with the actual one through universal polyscope UI. (**Beware** of frame conventions: [ROS](https://www.ros.org/reps/rep-0105.html) vs [Universal Robots](https://www.universal-robots.com/articles/ur-articles/parameters-for-calculations-of-kinematics-and-dynamics/))

fmauch, 2020-07-20:
Once you load the `robot_description` with a provided `calibration.yaml` file, the `robot_description` parameter on the parameter server will indeed contain a URDF WITH your calibrations applied.

You can, of course, also generate a static `.urdf` file as @cpt-yoshi explained above.

jbeck28, 2020-07-22:
Thank you both for the information, this has been very helpful!

fmauch, 2020-08-21:
Could this issue be closed?

fmauch, 2020-11-16:
Closing due to inactivity. Feel free to comment or reopen if anything new comes up.

closed 2020-11-16 by fmauch