UniversalRobots/Universal_Robots_ROS_Driver #229: Controlling Linear Speed

sapia123, 2020-07-22:
In URScript the linear endeffector speed can be controlled by using speedl(). But I couldn't find a rostopic to get this done. Is it planned to add a rostopic for controlling linear speed? Or does this contradict to ROS principles? I'm quite new to ROS - sorry if I misunderstand some concepts maybe.

For the moment I am controlling the linear speed by using the KDL library and the joint_group_vel_controller.

fmauch, 2020-07-23:
We are currently working on this. For direct linear control we will have something available probably in September.

saalimperator, 2020-12-09:
Are there any updates besides the [proposed interface](https://fzi-robot-interface-proposal.readthedocs.io/en/latest/proposal/index.html)?
Thank you for any information you are able to provide.

fmauch, 2020-12-09:
@stefanscherzinger Can you give any insight on the current status?

stefanscherzinger, 2020-12-10:
Development got delayed at little. Beta testing is scheduled to start mid-February and will be announced on all relevant channels for interested contributors. Earlier versions of specific features _might_ be available sooner, though.

saalimperator, 2020-12-11:
Thank you for the reply.
Since my thesis has to be handed in in January, I would like to implement the action client as soon as possible. However, I would not like to do the work twice. Will the proposed interface be implemented with at most minor changes or could it happen that the interface is completely reworked?

stefanscherzinger, 2020-12-11:
> Will the proposed interface be implemented with at most minor changes

Yes, we are currently implementing this interface.

> or could it happen that the interface is completely reworked?

The goal of this interface is to be suitable for most people. While it's unrealistic to meet everyone's needs, I personally think that the current version is built on sufficient consensus and will probably not change _drastically_ (I'll leave that term unspecified for now). However, it's still a community project and if broad usage of this interface uncovers new requirements, then sure should it change to meet _our_ community needs.

marbosjo, 2021-03-13:
@stefanscherzinger any news regarding this feature?

stefanscherzinger, 2021-03-13:
@marbosjo Although not officially announced yet, the code for beta testing is accessible [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/beta-testing#-beta-version-).

stefanscherzinger, 2021-08-09:
This feature will be integrated into the drivers mainline with #408.
The [twist_controller](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/staging/ur_robot_driver/config/ur5e_controllers.yaml#L141) will enable to drive the robot in Cartesian space with `geometry_msgs/TwistStamped` messages.

closed 2024-05-22 by VinDp