UniversalRobots/Universal_Robots_ROS_Driver #230: Unable to execute UR scripts

Pranav186, 2020-07-24:
# Summary
Unable to execute UR scripts

# Versions
Ubuntu 16.04
ROS Kinetic
UR10e robot

# Impact
Joint does not move on implementation of the script although there is no error on the terminal.

# Issue details

Following is the code for the publisher node

#!/usr/bin/env python

import rospy
from std_msgs.msg import String

rospy.init_node('vel_script_node')
pub=rospy.Publisher('/ur_hardware_interface/script_command', String, queue_size=10)
rate=rospy.Rate(10)
vel_script="speedj([0,0,0,0,0,-1.0],0.5,0.5)"

while not rospy.is_shutdown():
     pub.publish(vel_script)
     rate.sleep()

(It is indented properly.)

When i run this node, there is no error, but the joint does not move.
rqt_graph shows that there is communication between the two nodes over the ur_hardware_interface/script_command topic.
I have URCaps External control running.
I have set the remote control mode in the pendant.

Would like some help.

Thanks,
Pranav

Pranav186, 2020-07-28:
The issue was solved. In the github documentation, the remote mode is said to be enabled by going to Settings-Systems-Remote Control. That did not work for me. There was an icon on the top right corner which enabled the remote mode.

fmauch, 2020-07-30:
I'll leave this open as a reminder to update the documentation there.

In fact, one has to enable the remote-control mode(once)  in the settings before that icon appears.

fmauch, 2023-01-02:
On CB3 robots this should work out of the box, there exists no such thing as remote_control mode.

I just checked and at least with URSim this seems to work:

shell1:
```
rosrun ur_client_library start_ursim.sh -m ur5
```

then, wait for the robot to be booted. You can access Polyscope in the web browser on http://192.168.56.101:6080/vnc.html. When not using  URSim from the Docker image, make sure to switch on the robot.

shell2:
```
roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=192.168.56.101
```

shell3:
````
rviz -d $(rospack find ur_description)/cfg/view_robot.rviz
```

shell4:
```
rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'movel([0.5,0.5,0.5,3.14,-1.57,1.57])'"
```

This should make the robot move (visible in RViz and Polyscope) to this pose:

![image](https://user-images.githubusercontent.com/491645/210227609-7c4bd8df-c26d-49c6-8a0c-0a63dcb03ea5.png)

I currently cannot test this on a real robot, but I'd expect the same behavior. If not, have a look at the robot's log panel.

closed 2026-02-16 by urfeex