UniversalRobots/Universal_Robots_ROS_Driver #231: Strange behavior with /joint_group_vel_controller/command

Pranav186, 2020-07-26:
# Summary
Strange behavior with /joint_group_vel_controller/command 

# Versions
Ubuntu 16.04
ROS Kinetic
UR10e robot

# Impact
Unable to smoothly operate the joints using  /joint_group_vel_controller/command.  

# Issue details
I am using the /joint_group_vel_controller/command topic to send velocity commands.

Following is the procedure carried out to get the topic running-

```rosservice call /controller_manager/switch_controller "stop_controllers: ['scaled_pos_joint_traj_controller']"```

```rosservice call /controller_manager/switch_controller "start_controllers: ['joint_group_vel_controller']"```

The status of joint_group_vel_controller is "running".

I am using the following code for sending velocity commands-

#!/usr/bin/env python

import rospy
from std_msgs.msg import Float64MultiArray
from sensor_msgs.msg import JointState

rospy.init_node('velocity_node')
pub=rospy.Publisher('/joint_group_vel_controller/command', Float64MultiArray, queue_size=10)
msg=Float64MultiArray()
msg.data=[0,0,0,0,0,0.1]
msg.layout.data_offset=1

rate=rospy.Rate(10)

while not rospy.is_shutdown():
    pub.publish(msg)
    rate.sleep()

(It is indented properly)

1)After running this code, the joint moves with a weird noise.
2) Changing the speed in the code does not change the speed in the joint. 
3)Sometimes pausing and restarting the external control program on the pendant , and running the code again makes the joint move properly. Thus, I feel there is some glitch in the program and not a hardware issue. I tried the same for other joints and faced the same problem.

## Expected Behavior
Joint should run smoothly without noise. Changing velocity in the code should change actual joint velocity

## Actual Behavior
1)After running this code, the joint moves with a weird noise.
2) Changing the speed in the code does not change the speed in the joint. 
3)Sometimes pausing and restarting the external control program on the pendant , and running the code again makes the joint move properly. Thus, I feel there is some glitch in the program and not a hardware issue. I tried the same for other joints and faced the same problem.

## Other details
For stopping the robot, I change msg.data=[0,0,0,0,0,0.1] to msg.data=[0,0,0,0,0,0] .
Sometimes, I have also observed that the robot stops  immediately after making the change above without even saving and running this script using rosrun.

fmauch, 2020-07-30:
Thanks for reporting this. I don't have any spontaneous idea what could be the problem here, I will check again, when I am back in the office.

fmauch, 2020-09-07:
I cannot reproduce this on the UR16e I have currently access to. I modified your code in order to stop motion by pressing `ctrl + c` and to give the speed as cmd_line argument:

```python
#!/usr/bin/env python

import signal
import sys

import rospy
from std_msgs.msg import Float64MultiArray
from sensor_msgs.msg import JointState

class RobotMover():
    """Small node to move the robot"""

    def __init__(self, speed):
        super(RobotMover, self).__init__()

        rospy.init_node('velocity_node')
        self.pub = rospy.Publisher('/joint_group_vel_controller/command',
                                   Float64MultiArray, queue_size=10)
        self.msg = Float64MultiArray()
        rospy.loginfo("Speed is {}".format(speed))
        self.msg.data = [0, 0, 0, 0, 0, speed]
        self.msg.layout.data_offset = 1

    def spin_once(self):
        self.pub.publish(self.msg)

    def signal_handler(self, sig, frame):
        rospy.loginfo('You pressed Ctrl+C!')
        self.msg.data = [0, 0, 0, 0, 0, 0.0]
        self.pub.publish(self.msg)
        sys.exit(0)

if __name__ == "__main__":
    try:
        speed = float(sys.argv[1])
    except IndexError as err:
        rospy.logerr("No speed given.")
        sys.exit(1)
    except ValueError as err:
        rospy.logerr("No valid speed given")
        sys.exit(1)

    robot_mover = RobotMover(speed)
    rate = rospy.Rate(10)

    signal.signal(signal.SIGINT, robot_mover.signal_handler)

    while not rospy.is_shutdown():
        robot_mover.spin_once()
        rate.sleep()
```

Changing the speed definitely also works for me.

fmauch, 2020-09-07:
> Sometimes, I have also observed that the robot stops immediately after making the change above without even saving and running this script using rosrun.

This seems highly unlikely or that this is a coincidence.

Pranav186, 2020-09-07:
Ok, I will check again. You may close this if you want to.

fmauch, 2020-09-08:
Well, if the issue still exists for you, there's no reason to close this, then we should find out where that comes from.

fmauch, 2020-11-16:
@Pranav186 any updates on this?

Pranav186, 2020-11-16:
Hi @fmauch, unfortunately, I have not used the robot for a while and I do not have updates on this. Will try to take a look at it sometime soon.

fmauch, 2021-05-15:
@Pranav186 Have you been able to check this by now?

Pranav186, 2021-05-15:
Sorry, I am not working on the robot anymore and will not be able to check on this. You can close it if it looks like a coincidence and nobody has experienced similar behavior.

closed 2021-05-15 by fmauch

iiceberrg, 2022-09-17:
I try to use the modified code by @fmauch but the robot is not moving. Does anyone know why? Thank you.