UniversalRobots/Universal_Robots_ROS_Driver #233: Unable to bringup UR5

prajval10, 2020-07-28:
# Versions
- ROS Melodic
- Ubuntu 18.04
- UR5 robot 
   - UR software v3.8
- URcaps version 1.0.2
- Universal_Robots_ROS_Driver version used: commit https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/commit/e6b7941942e5d3d4319b688f7b98687196aa166a

# Impact
Unable to bringup the robot. I followed the methods as in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#setting-up-a-ur-robot-for-ur_robot_driver. I did not perform the [robot calibration](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#extract-calibration-information) as I wanted to check the quick start first. I have installed the externalcontorl URCap as specified. 

# Issue details
```[console]
SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 125
 * /hardware_interface/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 125
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /scaled_pos_joint_traj_controller/state_publish_rate: 125
 * /scaled_pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /scaled_pos_joint_traj_controller/type: position_controll...
[... 247 lines cut ...]
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Running the program from TP with the external control program node gives me the error:
```
Connection refused
The connection to the remote PC could not be established
```

## Setup
```
Robot IP: 172.22.22.10
Subnet mask: 255.255.255.0

ROS computer IP: 172.22.22.2
Subnet mask: 255.255.255.0
```

**Note: I am able to ping the robot ip from the computer**

## Project status at point of discovered
*When did you first observe the issue?*
 - *In first couple of tries?*

prajval10, 2020-08-07:
Friendly ping @gavanderhoorn, do you think updating the UR software version to newer version (such as v3.12 or v3.13) makes sense to solve this issue? I updated to v3.8 as the instructions specified that minimum requirement is v3.7

gavanderhoorn, 2020-08-07:
I am just a user, like you. @fmauch is the maintainer and developer here.

But looking at the console text you show, we see `Exception`s coming from the `ros_control` controller spawners, which are unable to communicate with the controller manager. I doubt this has anything to do with what version of software you run on your controller.

I would check whether there is a problem with your network configuration first. Perhaps a firewall is active? Do you have anything set for `ROS_IP`, `ROS_HOSTNAME` or `ROS_MASTER_URI` which is incorrect?

Do you have any outstanding updates? There was a problem with ABI incompatibilities with `ros_control` some time ago which resulted in similar tracebacks. Users with partially upgraded ROS installations were the first to notice this.

fmauch, 2020-08-07:
Sorry, that one slipped through my mails. Indeed it would be very interesting why the driver crashes.

Note that the launch files already support being run in debug mode:

`roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=172.22.22.10 debug:=true`

When it crashes, you will end up in a prompt like this:
```
Thread 1 "ur_robot_driver" received signal SIGINT, Interrupt.
[INFO] [1596792205.263327]: Stopping all controllers...
0x00007ffff6df8b28 in do_futex_wait.constprop () from /usr/lib/libpthread.so.0
(gdb)
```
There, type `bt`, press enter and post the output. It should look something like this:
```
(gdb) bt
#0  0x00007ffff6df8b28 in do_futex_wait.constprop () from /usr/lib/libpthread.so.0
#1  0x00007ffff6df8c53 in __new_sem_wait_slow.constprop.0 () from /usr/lib/libpthread.so.0
#2  0x00007ffff69a0d4b in moodycamel::spsc_sema::Semaphore::timedWait (this=0x555555675100, usecs=<optimized out>) at /home/mauch/robot_folders/checkout/ur_rtde_driver/catkin_ws/src/ur_rtde_driver/ur_robot_driver/include/ur_robot_driver/queue/atomicops.h:609
#3  moodycamel::spsc_sema::LightweightSemaphore::waitWithPartialSpinning (this=0x5555556750f8, timeout_usecs=<optimized out>) at /home/mauch/robot_folders/checkout/ur_rtde_driver/catkin_ws/src/ur_rtde_driver/ur_robot_driver/include/ur_robot_driver/queue/atomicops.h:667
#4  0x00007ffff699d2aa in moodycamel::spsc_sema::LightweightSemaphore::wait (this=0x555555675100, timeout_usecs=393) at /home/mauch/robot_folders/checkout/ur_rtde_driver/catkin_ws/src/ur_rtde_driver/ur_robot_driver/include/ur_robot_driver/queue/atomicops.h:710
#5  moodycamel::BlockingReaderWriterQueue<std::unique_ptr<ur_driver::rtde_interface::RTDEPackage, std::default_delete<ur_driver::rtde_interface::RTDEPackage> >, 512ul>::waitDequeTimed<std::unique_ptr<ur_driver::rtde_interface::RTDEPackage, std::default_delete<ur_driver::rtde_interface::RTDEPackage> > > (
    this=0x5555556750a8, result=std::unique_ptr<ur_driver::rtde_interface::RTDEPackage> = {...}, timeout_usecs=393) at /home/mauch/robot_folders/checkout/ur_rtde_driver/catkin_ws/src/ur_rtde_driver/ur_robot_driver/include/ur_robot_driver/queue/readerwriterqueue.h:790
#6  moodycamel::BlockingReaderWriterQueue<std::unique_ptr<ur_driver::rtde_interface::RTDEPackage, std::default_delete<ur_driver::rtde_interface::RTDEPackage> >, 512ul>::waitDequeTimed<std::unique_ptr<ur_driver::rtde_interface::RTDEPackage, std::default_delete<ur_driver::rtde_interface::RTDEPackage> >, long, std::ratio<1l, 1000l> > (this=0x5555556750a8, result=std::unique_ptr<ur_driver::rtde_interface::RTDEPackage> = {...}, timeout=...) at /home/mauch/robot_folders/checkout/ur_rtde_driver/catkin_ws/src/ur_rtde_driver/ur_robot_driver/include/ur_robot_driver/queue/readerwriterqueue.h:811
#7  ur_driver::comm::Pipeline<ur_driver::rtde_interface::RTDEPackage>::getLatestProduct (this=0x555555675068, product=std::unique_ptr<ur_driver::rtde_interface::RTDEPackage> = {...}, timeout=...)
    at /home/mauch/robot_folders/checkout/ur_rtde_driver/catkin_ws/src/ur_rtde_driver/ur_robot_driver/include/ur_robot_driver/comm/pipeline.h:331
#8  ur_driver::rtde_interface::RTDEClient::getDataPackage (this=0x555555674f50, timeout=...) at /home/mauch/robot_folders/checkout/ur_rtde_driver/catkin_ws/src/ur_rtde_driver/ur_robot_driver/src/rtde/rtde_client.cpp:337
#9  0x00007ffff69a3ed4 in ur_driver::UrDriver::getDataPackage (this=<optimized out>) at /home/mauch/robot_folders/checkout/ur_rtde_driver/catkin_ws/src/ur_rtde_driver/ur_robot_driver/src/ur/ur_driver.cpp:167
#10 0x00005555555e0460 in ur_driver::HardwareInterface::read (this=0x555555666f40, time=..., period=...) at /home/mauch/robot_folders/checkout/ur_rtde_driver/catkin_ws/src/ur_rtde_driver/ur_robot_driver/src/ros/hardware_interface.cpp:420
#11 0x00005555555fd8fc in main (argc=<optimized out>, argv=<optimized out>) at /home/mauch/robot_folders/checkout/ur_rtde_driver/catkin_ws/src/ur_rtde_driver/ur_robot_driver/src/ros/hardware_interface_node.cpp:136
```

prajval10, 2020-08-13:
> Do you have any outstanding updates? There was a problem with ABI incompatibilities with ros_control some time ago which resulted in similar tracebacks. Users with partially upgraded ROS installations were the first to notice this.

This seemed to do the trick. I had some outstanding updates for `ros_control` which I was not aware of, simply upgrading them seemed to resolve the issue. The robot is up and running

```
[ INFO] [1597306909.148381869]: Service available.
[ INFO] [1597306909.368412327]: Robot's safety mode is now NORMAL
[ INFO] [1597306909.371230444]: Robot mode is now RUNNING
[INFO] [1597306909.421606]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1597306909.422781]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1597306909.429207]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1597306909.431950]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1597306909.437604]: Loading controller: joint_state_controller
[INFO] [1597306909.440262]: Loading controller: pos_joint_traj_controller
[INFO] [1597306909.467490]: Loading controller: scaled_pos_joint_traj_controller
[INFO] [1597306909.546512]: Loading controller: joint_group_vel_controller
[INFO] [1597306909.586707]: Loading controller: speed_scaling_state_controller
[INFO] [1597306909.594222]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[INFO] [1597306909.602237]: Loading controller: force_torque_sensor_controller
[INFO] [1597306909.610370]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1597306909.618541]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[ INFO] [1597306990.884550153]: Robot requested program
[ INFO] [1597306990.884748477]: Sent program to robot
[ INFO] [1597306992.287477760]: Robot ready to receive control commands.
```

Thanks @gavanderhoorn and @fmauch. Closing.

closed 2020-08-13 by prajval10