UniversalRobots/Universal_Robots_ROS_Driver #234: Switch between ROS driver and teach pendant

tonkei0361, 2020-07-29:
# Summary

I found an old post below discussing how people wanted to switch the method of controlling a robot between ROS driver and teach pendant. Here is the post:

https://github.com/ros-industrial/universal_robot/issues/54

The driver in this repo does not seem to allow it. While the external_control script is running, any control commands from the teach pendant will either be blocked by the external_control or stop the external_control. They also mentioned something like 'uploading a driver script from python driver'. Here I think the driver script they mention is similar to external_control script but from what I noticed, as soon as external_control script is interrupted, the UR driver running on the computer cannot  resume/restart the external_control script unless the robot is set to remote control mode. However, in remote control mode, the teach pendant is disabled... Am I correct on this part?

One workaround they mentioned is to call the URScript method `teachmode()`. Is this supported in this driver?

# Versions
 - ROS Driver version: Melodic
 - Affected Robot Software Version(s): 5.8
 - Affected Robot Hardware Version(s): UR5e

fmauch, 2020-07-30:
First of all, please note, that this post is very old and does not refer to this driver by any means.

The behavior how the driver is currently working is intended for multiple reasons:
 - For safe robot operation, there has to be one single instance in control over the robot the whole time. By default, this is the teach pendant. When putting the robot into remote-control mode, it is the external control structure controlling the robot.
 - We wanted to achieve transparency when ROS-control is actually active on the robot and when not. This was from our experience one major drawback with previous driver implementations that after a manual interaction on the teach pendant, the state of the ROS driver was never really clear.

For those reasons the structure is as follows:
 - As long as ROS-Control is active, the program will be running on the teach pendant (TP), clearly indicating to the user, that there is something running.
 - The ROS driver outputs a clear message `Robot ready to receive control commands.` when ROS control is usable, while it will show a connection drop message, once the program stops running and the corresponding ROS controllers will be put into a stop mode, where they won't accept new commands anymore.Continuing the program on the TP will restart the running controllers and make the robot accept commands again.

As a consequence, this means that you can simply stop the program running on the TP by pressing the "stop" button on the TP, do your manual interaction with the robot (use the motion buttons, use the teach button, whatever) and then press "play" again, to re-enable ROS control.

fmauch, 2020-08-04:
> My understanding is that while the driver is running, Digital Inputs are not usable. 

What brings you to that conclusion? I/Os are usable as normal and their state is published by the robot driver on `/ur_hardware_interface/io_states`. For example, you can see the safeguard_stop input being active in our [release video](https://youtu.be/EfEfkXUJwP8?t=86) using a laser scanner connected to the robot's configurable inputs.

When mapping a URScript function to a digital input as explained in the article you linked, this should be working as expected. As this will effectively run a new program, it will stop the external control program as soon as the freedrive button is pressed. After moving the robot around, you can press "play" on the TP again to restart ROS-control (or call the [respective dashboard service](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#dashboardplay-std_srvstrigger), if the robot is in remote_control mode).

Note that for the normal freedrive button on the TP it seems to be necessary to stop a running program first (e.g. by pressing the "stop" button or manually driving the robot using the arrows on the TP). It might be the same using an external freedrive button. It would be nice if you shared your experience here, once you implemented this.

> Also, maybe I can do the same with `/ur_hardware_interface/script_command`? Just wanted to double check if I am right on this. Thank you very much!

You can also activate the freedrive mode using a script_command, but this will only be active as long as the script keeps running. Therefore, you would have to send a script snippet that runs longer after activating freedrive mode. This doesn't seem very practical to me.

stefanscherzinger, 2021-09-01:
@tonkei0361 Any news you want to share?

closed 2023-04-07 by github-actions[bot]