UniversalRobots/Universal_Robots_ROS_Driver #235: Driver does not build on Melodic due to "SetPayload" service

stefanz-tri, 2020-07-29:
It seems like the driver does not build on melodic (and kinetic) currently because of https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/50 . When using the universal_robot metapackage, the "SetPayload" service is missing the "center_of_gravity" member. When using the standalone ur_msgs package, it does not build because the "payload" member was renamed to "mass". Or is this already solved in a branch?

fmauch, 2020-07-30:
Did you use my fork of the `universal_robot` repository, [as explained in the README](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building)?

stefanz-tri, 2020-07-30:
No I actually used the main repository. It is a bit confusing that now three versions of ur_msgs exist. But that should solve the issue, thanks.

closed 2020-07-30 by stefanz-tri

anshulpaigwar, 2020-11-16:
I am facing the same issue and could not figure out which fork I must use for Melodic??

The current version of the ur_msg repository has a variable name Mass
https://github.com/ros-industrial/ur_msgs

And the ros driver hardware interfaces file is still using payload

fmauch, 2020-11-16:
@anshulpaigwar Please follow the [instructions in the README file](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building)

suc1, 2021-10-04:
My environment is:
[Melodic (bionic 18.05.5)](https://rosi-images.datasys.swri.edu/) ; Virtual Box 6.1

Following README file but still got errors:
[image]

What's wrong?

Thanks

suc1, 2021-10-05:
I only followed [the instructions](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver)
and build successfully.

I think my previous trouble was caused in [universal_robotTutorials](http://wiki.ros.org/universal_robot/Tutorials/Getting%20Started%20with%20a%20Universal%20Robot%20and%20ROS-Industrial
)
`sudo apt-get install ros-kinetic-universal-robots`

fmauch, 2021-10-05:
@suc1 please do not hijack an old (and closed) issue for your problems. You can link to related issues, by simply writing `#235` inside the issue.

Before you open a new issue: Please note that ros kinetic is no longer supported.