UniversalRobots/Universal_Robots_ROS_Driver #236: How to subscirbe robot states (joint/end effector position/velocity) and publish a trajectory command

ceciliaxu1, 2020-08-03:
# Summary
I am new in ROS. And I wonder how to subscirbe robot states (joint/end effector position/velocity)?
I am using "kinematic_model->getGlobalLinkTransform("ee_link")" in my ros to get end effector's state, but results of two times are different. And I wonder there are easier way to get this.
The other question is how to publish a trajectory and control UR in ROS?
Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*

# Versions
 - ROS Driver version:
 - Affected Robot Software Version(s):
 - Affected Robot Hardware Version(s):
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):

# Impact
*What is the impact of the issue. Try to describe it well enough, in order for us to prioritise the issues.*

# Issue details
*This part is important in order to clarify the severity of the issue.*

## Use Case and Setup
*What does the setup look like and what are the objectives?*

## Project status at point of discovered
*When did you first observe the issue?*
 - *Before I started?*
 - *In first couple of tries?*
 - *In normal use?*
 - *etc...*

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

fmauch, 2020-08-04:
For questions how to *use* this driver, please refer to ROS answers. This issue board is for tracking problems inside this driver. Especially, if you are new to ROS, it is generally a good idea to ask other ROS users rather than individual package developers.

You might want to read up on [using TF ](http://wiki.ros.org/tf/Tutorials/Introduction%20to%20tf) and [joint_state msgs](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/JointState.html) which are published by the joint_state_controller on `/joint_states` by default.

fmauch, 2020-08-04:
On information how to send trajectory commands, please refer to the [joint_trajectory_controller documentation](http://wiki.ros.org/joint_trajectory_controller).

closed 2020-08-04 by fmauch

gavanderhoorn, 2020-08-04:
Perhaps https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/122#issuecomment-591861319 and #162 can provide some info here?

ceciliaxu1, 2020-08-12:
Hi,
Thank you for your reply. The problems have been solved now.

I have another question about the universal_robot_ros_driver. I have read the docs about real-time kernal but still confuced. Can improve the real-time feature of the system? And is it effective for a UR5 not UR5e.

Cause I get the info
"[ WARN] [1597237012.671293627]: No realtime capabilities found. Consider using a realtime system for better performance"
"[ERROR] [1597240698.444282296]: Sending data through socket failed."

Thank you again for you help~~

fmauch, 2020-08-13:
See #243 about the real-time reasoning.

Regarding your socket-send failure: Please open a new issue for that giving a FULL log output of the driver. And please use a [code block](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) to post the log output.