UniversalRobots/Universal_Robots_ROS_Driver #237: UR10e  - huge lag when programs loaded/played from the Ubuntu PC

AsadT, 2020-08-03:
I want to load and run programs that I have on the UR. The ROS PC has a perfect connection to the UR. When the load command is sent the programs gets loaded. When we run it. The UR gives a popup on the polyscope that says "starting program". Before we installed the external control URcap and the gripper (schunck EGH-80) URcap the programs would run immediately after pressing play. Since these URcaps the delay is huge. The ROS driver times out after one second. So we increased the timeout period. But that is no solution as simple motion programs are taking as much as a few minutes to load and play. On the DoF forum I saw two similar queries with ROS and Gripper URcaps. But they arent answered yet. Any body here has an idea how I can get the execution in a respectable time. Thanks

fmauch, 2020-08-04:
A bit more context would be nice to have.
First of all:
 - Is the external_control URCap installed (and setup properly!)?
 - Without knowing the script code that you are trying to execute, this is hard to debug. Please post the generated script code of the program in question. See [this post](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/74#issuecomment-571949829) how to get those.
 - What other URCaps are installed on the system?
 - What robot version are you actually using?

The default issue template is there for a reason. It helps us getting the necessary information to track down problems more efficiently. Please use that.

AsadT, 2020-08-04:
Sorry for not following the template. In any future post, I would follow. 

- The external URCap is installed properly and runs fine. I have checked it by sending velocities from Ubuntu PC. From ROS the driver also loads, stops and plays the program inside the UR. 

- It doesn't matter which program I am trying to load and play. The problem remains the same that It takes a long time to play with a pop up saying "starting program". The test program in question is a simple MoveJ from one waypoint to another. If you want I can send the script code. But there is really nothing more in it than that. The issues are exactly like the one reported [here on the DOF forum](https://dof.robotiq.com/discussion/726/lag-time-between-pressing-play-on-ur-and-program-start-with-ucg-1-1-1) 

- The only other URcap is Schunk EGH gripper. It is for the installed collaborative gripper from Schunk

- Its a UR-10e with the software 5.4.2.7XXX

fmauch, 2020-08-04:
> * It doesn't matter which program I am trying to load and play. The problem remains the same that It takes a long time to play with a pop up saying "starting program". The test program in question [...]

Without the **generated** script code there is not much I could say.

fmauch, 2020-08-04:
> * The only other URcap is Schunk EGH gripper. It is for the installed collaborative gripper from Schunk

Does the problem still exist if this URCap is uninstalled?

gavanderhoorn, 2020-08-04:
@fmauch: the posts by the user `Stefan` on the Robotiq forum ([here](https://dof.robotiq.com/discussion/comment/5767/#Comment_5767)) are interesting:

> I just wanted to let you know that I solved my problem. It turned out that the firewall UFW (Uncomplicated Firewall) was installed and active on my Ubuntu PC. I just needed to disable the firewall to get it working (sudo ufw disable).
> 
> Now I can start my external control program and the connection to my ROS PC is established instantly. I can also start a ROS independent program without any lag. That`s great but it is also very strange that my firewall had such a big effect on the UR.

Is this connected/related to UniversalRobots/Universal_Robots_ExternalControl_URCap#5 (in that initialisation of the ExtControl urcap is perhaps not in the right place)?

Connections needing to time-out could be causing the perceived "lag".

AsadT, 2020-08-12:
> Without the **generated** script code there is not much I could say.

Here is the zip folder including the log that you asked for. The zip folder also contains the programs we are trying to load and play via the ROS package. The programs are loaded in this order cam01, cam02, cam03 and then calibrate. Between the programs a pop up appear about "starting programing". That can last from anywhere between a few seconds to 2 minutes. 

if that happens ROS breaks the connection. We have increased the time out for the connection but this is no solution as we would like to have seamless loading and playing 
[UR10e_files.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/5062467/UR10e_files.zip)

AsadT, 2020-08-13:
@fmauch: Do you find anything in the files I shared to figure out what is going on?

AsadT, 2020-08-13:
We removed the gripper URCap - and the problem remained even manually on the polyscope there was a long delay in pressing play and the robot actually starting. 

We removed the external control URCap - the problem disappeared. We can load manually and as soon as the play button is pressed the response is instant.

fmauch, 2020-08-13:
The script code inside e.g.  `mobot_engraver_cam01.script` shows the following lines:

```python
  # begin: URCap Installation Node
  #   Source: External Control, 1.0.1, FZI Research Center for Information Technology
  #   Type: External Control
  # end: URCap Installation Node
```

This is unusual as I would expect that between the line `#   Type: External Control` and the line `# end: URCap Installation Node` would be either the script code as sent by the ROS machine or the code (given that you are using version 1.0.2 of the URCap, which you didn't say so far.)

```python
popup("The connection to the remote PC could not be established. ", "connection refused", False, True, blocking=True)
sync();
```

As @gavanderhoorn mentioned earlier, it looks like this can happen, when a firewall is active on the ROS machine.

Basically, [this function](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap/blob/master/src/main/java/com/fzi/externalcontrol/impl/RequestProgram.java#L48) gets executed, which seems to throw another exception than `IOException`.

fmauch, 2020-08-13:
With ufw installed and enabled with a "Deny all" setup I can see the same behavior on my machine.

From your OP I was assuming that you can actually get and run the program code, probably I was misunderstanding that.

gavanderhoorn, 2020-08-13:
@fmauch: is there any support for specifying connection timeouts in URScript? I haven't checked, could be an RTFM.

fmauch, 2020-08-13:
> @fmauch: is there any support for specifying connection timeouts in URScript? I haven't checked, could be an RTFM.

It's actually not URScript, but UR+ Java SDK, but yes, I am currently reading the API docs ;-)

AsadT, 2020-08-13:
Thanks a lot. It's solved but the problem and the solution make no sense. 

It turned out that the IP set for the external control was 192.168.56.XX

While the ROS PC was communicating with 192.168.50.XX. All other components like Gripper were on the 50.XX subnet as well.

When we detected that and put the correct IP. The pop ups disappeared.

What is highly surprising is that with a wrong IP, that is 56.XX, how were we able to load and run the programs .albeit with delay. The connection should have been refused.

gavanderhoorn, 2020-08-13:
> > @fmauch: is there any support for specifying connection timeouts in URScript? I haven't checked, could be an RTFM.
> 
> It's actually not URScript, but UR+ Java SDK, but yes, I am currently reading the API docs ;-)

Ah, yes. I get confused sometimes ;)

fmauch, 2020-08-13:
> > > @fmauch: is there any support for specifying connection timeouts in URScript? I haven't checked, could be an RTFM.
> > 
> > 
> > It's actually not URScript, but UR+ Java SDK, but yes, I am currently reading the API docs ;-)
> 
> Ah, yes. I get confused sometimes ;)

And I was actually wrong, it's not the UR+ SDK in that case, but plain Java: https://docs.oracle.com/javase/7/docs/api/java/net/Socket.html

PR #245 (2020-09-14): Updated externalcontrol to v1.0.3
See the urcap's release notes [here](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap/releases/tag/v1.0.3).

I'll leave this unmerged for a couple of days until I have tested it more thoroughly.

Closes #237
closed 2020-09-14 by fmauch