UniversalRobots/Universal_Robots_ROS_Driver #239: tool_digital_output_mask is currently controlled by another RTDE client

SmirkCao, 2020-08-05:
# Summary

UR5+RG2
I have installed onrobot urcap to control rg2 and got the error "tool_digital_output_mask is currently controlled by another RTDE client" when i launch the ur5 driver.
If I remove onrobot urcap,  the driver will start normally.

may I use ROS and onrobot urcap together?

# Versions
 - ROS Driver version: Kinetic
 - Affected Robot Software Version(s): 3.13.0
 - Affected Robot Hardware Version(s):
 - Robot Serial Number:
 - UR+ product(s) installed: OnRobot 5.1.6
 - URCaps Software version(s): External Control 1.0.1

SmirkCao, 2020-08-05:
I have edited the file ur_robot_driver/resources/rtde_input_recipe.txt
remove tool_digital_output* and relaunch, it works.

closed 2020-08-05 by SmirkCao

gavanderhoorn, 2020-08-05:
This reads like a similar issue as #204 (as in: in the same category).

ehannigan, 2021-06-25:
Had the same issue and your solution fixed it! Thanks. Do you know why this works? What does the rtde_input_recipe.txt file do?

fmauch, 2021-06-28:
The file effectively reserves input fields of the RTDE interface for this driver. See the [upstream documentation](https://www.universal-robots.com/articles/ur/interface-communication/real-time-data-exchange-rtde-guide/) for details.

If there are two RTDE clients claiming the same input fields (e.g. a URCap controlling a device through RTDE as in the OP) and the ROS driver, this does not work. By commenting out those, the driver doesn't claim those resources, but is also not able to control those. In the end, you won't be able to set the `tool_digital_output*` fields from ROS.