UniversalRobots/Universal_Robots_ROS_Driver #243: Some problem about the RealTime-enabled

TFLQW, 2020-08-12:
# Summary

*I cannot understand why we should enable the RealTime-enabled because I have found that the  communication frequency of different posts are set constant such as port 30002 is 10Hz, port 30003 and 30004 is 125Hz, the robot I am using is UR10 and UR5, although I have enabled the real time function, I think I cannot break through the port's frequency limit. So I am confused by the Real-Time function, could anyone can help me answer this question?*

# Versions
 - ROS Driver version:  Ubuntu 18.04
 - Affected Robot Software Version(s): 3.8
 - Affected Robot Hardware Version(s): CB3
 - Robot Serial Number: UR10
 - UR+ product(s) installed:
 - URCaps Software version(s):

fmauch, 2020-08-12:
To ensure smooth robot control it is important that the robot receives a motion command in each control cycle. The control loop runs at 125Hz for CB3 series, while at 500Hz for the eSeries.

The problems on a non-realtime scheduling is that cycle times on the ROS machine become much less deterministic, as other processes might kick in and delay the actual driver execution. This gets to a point where some control cycles end up being much longer than the required 8ms (as with the 125Hz for CB3 robots).

This has negative side effects such as
 - Data buffers might overflow meaning data coming from the robot will be lost
 - The robot will not receive new setpoints in time resulting in a deviation from the desired path or jerky motions in the worst case

Although, you might get away without using a realtime kernel (I personally use a non-realtime system for development) we recommend using a real-time enabled system in order to make sure the driver runs smoothly.

TFLQW, 2020-08-12:
@fmauch Thanks for your answer, you mean that no matter CB3 Series or eSeries, the realtime kernel will improve the control performance, but the CB3 Series with non-realtime system and with realtime system performance gap is not so significant, it is that right?

fmauch, 2020-08-12:
> @fmauch Thanks for your answer, you mean that no matter CB3 Series or eSeries, the realtime kernel will improve the control performance,

depending on your definition of "control performance", yes.

> but the CB3 Series with non-realtime system and with realtime system performance gap is not so significant, it is that right?

That would be my assumtion. We've only did a short 1-minute test with an e-series robot a while ago, where we noticed, that without a real-time kernel cycle rates spiked up into the 16 ms section.This was not a scientific reliable test, but it gives the idea of the problem. With a real-time kernel cycle times in a ten minutes test stayed below 2.4 ms.

I can and will not give you any advice, in which situation it might be suitable to use a non-real-time kernel. We suggest to use a real-time kernel. If you don't want to set it up, go for it, but be aware that it *could* introduce those negative side effects mentioned earlier.

TFLQW, 2020-08-12:
@fmauch Ok, I have understood！

closed 2020-08-12 by TFLQW

gavanderhoorn, 2020-08-12:
> The problems on a non-realtime scheduling is that cycle times on the ROS machine become much less deterministic

Just a minor nuance/clarification here: the problem with determinism is not a "ROS problem" per se. No system will be deterministic without proper real-time support in the OS kernel, userland and the programs written using that infrastructure. ROS does not change anything there.

In the context of the question posted by @TFLQW, obviously "the ROS machine" refers to the computer which is running the driver node, but I wanted to add this clarification, as this (ie: real-time execution of nodes) is a topic which comes up often and there is a lot of confusion about the whats and the whys.