UniversalRobots/Universal_Robots_ROS_Driver #244: connection refused

IndustrialEngStudent, 2020-08-13:
# Summary

Connection to remote PC could not be established

*Introduction to the issue*
I am following these instructions: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md
At the last step when I try to run the program I get a pop up saying: "The connection to the remote PC could not be established"
What am I doing wrong?

# Versions
 - ROS Driver version: Melodic
 - CB3 controller 
 - UR software 3.12.0.90886
 - URCaps Software version(s):URcap External control version 1.0.1

# Issue details

```
(base) laurah@laurah-HP-EliteBook-840-G1:~/catkin_ws$ ifconfig -a
enp0s25: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.200  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::c64b:bf24:c033:3280  prefixlen 64  scopeid 0x20<link>
        ether 28:80:23:d6:28:d5  txqueuelen 1000  (Ethernet)
        RX packets 2899  bytes 189204 (189.2 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 992  bytes 127630 (127.6 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 20  memory 0xd0700000-d0720000  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 8823  bytes 753922 (753.9 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 8823  bytes 753922 (753.9 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlo1: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 80:86:f2:69:4d:08  txqueuelen 1000  (Ethernet)
        RX packets 70476  bytes 68420709 (68.4 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 34546  bytes 24819433 (24.8 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

``` 
(base) laurah@laurah-HP-EliteBook-840-G1:~/catkin_ws$ ping 192.168.1.100
PING 192.168.1.100 (192.168.1.100) 56(84) bytes of data.
64 bytes from 192.168.1.100: icmp_seq=1 ttl=64 time=0.267 ms
64 bytes from 192.168.1.100: icmp_seq=2 ttl=64 time=0.319 ms
64 bytes from 192.168.1.100: icmp_seq=3 ttl=64 time=0.270 ms
^C
--- 192.168.1.100 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2041ms
rtt min/avg/max/mdev = 0.267/0.285/0.319/0.027 ms
```

[image]
[image]

[image]

fmauch, 2020-08-13:
Did you start the ROS driver prior to starting the program on the panel?

If so, is there maybe some firewall involved? You can check with `sudo ufw status`

Also, what's the output of `-atnp | grep 5000`? It should show port 50001 and 50002 with state `LISTEN`:

```
$ ss -atnp | grep 5000
LISTEN     0      1                                        0.0.0.0:50001                             0.0.0.0:*     users:(("ur_robot_driver",pid=1396538,fd=18))
LISTEN     0      1                                        0.0.0.0:50002                             0.0.0.0:*     users:(("ur_robot_driver",pid=1396538,fd=16))

fmauch, 2020-08-14:
With version 1.0.3 of the URCap the popup got added a reason why connection failed. You could try it out and download it [here](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap/releases/tag/v1.0.3)

IndustrialEngStudent, 2020-08-14:
> Did you start the ROS driver prior to starting the program on the panel?
No, sorry for the inconvenience, I misunderstood the instructions. I thought I needed to get connection via the tablet first and afterwards start the ROS driver.
I started the ROS driver and I do not get the pop up now. Thank you for your help!

closed 2020-08-14 by IndustrialEngStudent