UniversalRobots/Universal_Robots_ROS_Driver #247: The difference between the /scaled_pos_traj_controller and /pos_traj_controller

TFLQW, 2020-08-19:
# Summary

Could tell me the difference between the controllers of   /scaled_pos_traj_controller and /pos_traj_controller

*Introduction to the issue*
I have tried to use the  controllers of **/scaled_pos_traj_controller and /pos_traj_controller**, when I sent the goal position of the joints to the robot, I found no diiference between them. So could you explain why you create these two controllers maybe have  the same function.

# Versions
 - ROS Driver version:
 - Affected Robot Software Version(s): CB3
 - Affected Robot Hardware Version(s): UR10
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):

# Issue details
*This part is important in order to clarify the severity of the issue.*

fmauch, 2020-08-19:
See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_controllers/README.md for a quick answer. This description could probably be improved, though.

fmauch, 2020-08-21:
@TFLQW I've improved the documentation a bit: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/controller_doc/ur_controllers/README.md

Could you please comment whether this leaves any open questions?

PR #250 (2020-09-14): Added explanation of scaled trajectory controller
This resolves #247