UniversalRobots/Universal_Robots_ROS_Driver #249: AttributeError: 'RobotStateRT' object has no attribute 'q_actual'

gachiemchiep, 2020-08-20:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*

Hello. I follow your instruction, but can't get the ur5 to work for this driver. 
The robot complained that incoming message has wrong length, and shut everything down.

```bash
roslaunch ur_bringup ur5_bringup.launch robot_ip:=192.168.5.102
```

Logs 

```bash
SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_ip_address: 192.168.5.102
 * /robot_reverse_port: 50001
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /tf2_buffer_server/buffer_size: 120.0
 * /ur_driver/max_payload: 10.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    tf2_buffer_server (tf2_ros/buffer_server)
    ur_driver (ur_driver/driver.py)

auto-starting new master
process[master]: started with pid [407580]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ae0711ac-e2d1-11ea-b90b-54bf64969598
process[rosout-1]: started with pid [407593]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [407597]
process[ur_driver-3]: started with pid [407611]
process[tf2_buffer_server-4]: started with pid [407613]
Setting prefix to 
[WARN] [1597920080.448697]: No calibration offset for joint "shoulder_pan_joint"
[WARN] [1597920080.449091]: No calibration offset for joint "shoulder_lift_joint"
[WARN] [1597920080.449352]: No calibration offset for joint "elbow_joint"
[WARN] [1597920080.449593]: No calibration offset for joint "wrist_1_joint"
[WARN] [1597920080.449768]: No calibration offset for joint "wrist_2_joint"
[WARN] [1597920080.449937]: No calibration offset for joint "wrist_3_joint"
[INFO] [1597920080.450208]: No calibration offsets loaded from urdf
[INFO] [1597920080.451856]: Max velocity accepted by ur_driver: 10.0 [rad/s]
[INFO] [1597920080.454736]: Bounds for Payload: [0.0, 10.0]
Disconnected.  Reconnecting
[INFO] [1597920080.514964]: Programming the robot
[INFO] [1597920080.519248]: Programming the robot at 192.168.5.102
RobotStateRT has wrong length: 1116
Exception in thread URConnectionRT:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/jil/workspace/catkin_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 372, in __run
    self.__on_packet(packet)
  File "/home/jil/workspace/catkin_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 338, in __on_packet
    for i, q in enumerate(stateRT.q_actual):
AttributeError: 'RobotStateRT' object has no attribute 'q_actual'

Likely a syntax error:
18-05-2020, 15:25:15trol
Exception in thread URConnection:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/jil/workspace/catkin_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 274, in __run
    self.__on_packet(packet)
  File "/home/jil/workspace/catkin_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/driver.py", line 186, in __on_packet
    state = RobotState.unpack(buf)
  File "/home/jil/workspace/catkin_ws/src/fmauch_universal_robot/ur_driver/src/ur_driver/deserialize.py", line 324, in unpack
    raise Exception("Fatal error when unpacking RobotState packet")
Exception: Fatal error when unpacking RobotState packet

Handling a request
[INFO] [1597920081.233831]: Robot connected
[2020-08-20 19:41:21.240707] Out: hello
^C[tf2_buffer_server-4] killing on exit

```

# Versions
 - ROS Driver version: Ubuntu 16.04, ros-kinetics
 - Affected Robot Software Version(s): 3.13
 - Affected Robot Hardware Version(s): ur5
 - Robot Serial Number: 
 - UR+ product(s) installed: 
 - URCaps Software version(s): 1.0.2

gavanderhoorn, 2020-08-20:
> ```
> roslaunch ur_bringup ur5_bringup.launch robot_ip:=192.168.5.102
> ```

you are not using the `Universal_Robots_ROS_Driver` `.launch` files, but the old ones in `ur_bringup`.

That's the cause of your problems.

gachiemchiep, 2020-08-20:
@gavanderhoorn 

Thanks . You have save me 2 days.

In [https://github.com/UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) the command is right.

    $ roslaunch ur_robot_driver <robot_type>_bringup.launch robot_ip:=192.168.56.101

But in the [https://github.com/fmauch/universal_robot](https://github.com/fmauch/universal_robot) it's still in the old format. 

    $ roslaunch ur_bringup ur5_bringup.launch limited:=true robot_ip:=IP_OF_THE_ROBOT [reverse_port:=REVERSE_PORT]

closed 2020-08-20 by gachiemchiep