UniversalRobots/Universal_Robots_ROS_Driver #254: Noetic Release

tylerjw, 2020-09-03:
# Summary

Once the issues with the Melodic release are resolved could we get a Noetic release of this too?

fmauch, 2020-09-04:
For noetic the same requirements as for the melodic release are valid. This cannot be released as a binary package as long as upstream dependencies aren't released. As soon as they are met, we will be happy to make a noetic release.

Until that happens, this package should be building on noetic from source. If not, please open a specific issue with the problems encountered.

stefanscherzinger, 2021-09-01:
@fmauch Since upstream dependencies are released now for both *Melodic* and *Noetic*, we could reconsider this.

fmauch, 2021-09-01:
Description is still missing, so no.

miladasgari178, 2023-09-22:
Hello, I am trying to use this tutorial "https://github.com/UniversalRobots/Universal_Robots_ROS_Driver" for creating UR5 Driver ROS package, for ROS noetic, but when I want to go through steps, I encounter problem 
what can I do?
can I use this driver in ROS noetic?

fmauch, 2023-09-25:
We are currently overhauling the README regarding this (#655). Yes, you can use the driver with MoveIt and it is released, so you can simply install it using `sudo apt install ros-noetic-ur-robot-driver`.

closed 2023-09-25 by fmauch