UniversalRobots/Universal_Robots_ROS_Driver #256: UR Polyscope program stuck in "Starting program"

martamillet, 2020-09-07:
# Summary

*Introduction to the issue*
When the ROS driver is launched and we start the execution of a UR Polyscope program (any program, we also tried a simple program with just a wait command) the HMI gets stucked with the pop-up "Starting program... please wait"
# Versions
 - ROS Driver version: up to date 7/09/2022
 - Affected Robot Software Version(s):  5.9
 - Affected Robot Hardware Version(s): UR10e
 - Robot Serial Number: 20205090026
 - UR+ product(s) installed:none
 - URCaps Software version(s):

# Impact
HIGH
we are unable to use the robot normally to execute polyscope programs.

gavanderhoorn, 2020-09-07:
Likely a duplicate of #237.

gavanderhoorn, 2020-09-07:
Might also want to take a look at #245 and use the urcap from that PR to see whether you get the error popup sooner.

martamillet, 2020-09-07:
It is solved when deleting the External Control URCap. 
But as soon as this URCap is installed it stops working (tried with the latest version of everything, repository and URCap)

gavanderhoorn, 2020-09-07:
Have you read the comments in #237?

martamillet, 2020-09-07:
You were right, even if you don't use the External Control URCap in your program, but you have it installed, you should modify the IP to make it right. That solved the issue, once we changed the IP in the installation file it worked.

gavanderhoorn, 2020-09-07:
Yes, this is not ideal right now.

It's a known issue, and is being worked on, but right now you have to make sure you've configured the correct IP.

closed 2020-09-07 by martamillet

fmauch, 2020-09-07:
@martamillet did you try the URCap from #245 whether that gives a clearer feedback?

martamillet, 2020-09-07:
With that version the issue apparently it doesn't happen