UniversalRobots/Universal_Robots_ROS_Driver #257: Check if URCap/ROS_external_control is running on the robot

felixvd, 2020-09-08:
It would be helpful to know if the URCap/`ROS_external_control` is running on the robot, i.e. that it is ready to receive joint trajectories through ROS. Currently, the only method I see is checking if a program is running and that the name matches `ROS_external_control.urp`, but this is exceedingly slow. Is there a way to check this that I missed?

If there is none, I would publish `reverse_interface_active_` as a `std_msgs::Bool`, since [it is true while the robot is connected](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/ur/ur_driver.cpp#L215). I can file a PR if it's welcome.

# Versions
 - ROS Driver version: master branch (on commit 3ce362, but it doesn't look like there was a change related to this)

fmauch, 2020-09-08:
This is already implemented, as documented in the [ROS_INTERFACE.md](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#robot_program_running-std_msgsbool).

felixvd, 2020-09-08:
Thanks! I wouldn't have guessed that that's specific to the external_control program. I thought it is true when any program is running on the robot, just like the `isProgramRunning` service. Renaming it seems good to me, but this will work for me.

closed 2020-09-08 by felixvd