UniversalRobots/Universal_Robots_ROS_Driver #258: Dashboard client does not survive minor errors

felixvd, 2020-09-08:
The dashboard client disconnects when it encounters an error / does not receive an answer. I think that it should not disconnect, but reset the socket connection instead.

For me, 95% of the time, the error occurs because I left the teaching pendant in Local mode. However, instead of changing to Remote Control and continuing, the whole launch file needs to be restarted, only for the dashboard client. This seems unnecessary.

If there is no particular reason for this behavior I am happy to file a PR.

fmauch, 2020-09-08:
This could definitely be improved, I would have to take a deeper look into the code to give a proper answer.

However, you don't have to restart the whole driver, there are the services [`dashboard/quit*](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#robot_program_running-std_msgsbool) and [`dashboard/connect`](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#dashboardconnect-std_srvstrigger) to disconnect and connect to the dashboard server.

When trying to make a call that requires remote control enabled, the driver will tell you that in the output and that you should reconnect the dashboard server. If you could provide a specific (reproductible) use case where you think, the behavior should be changed, we can try to find a viable solution for that :-)

felixvd, 2020-09-08:
Is there a reason not to quit and reconnect immediately after the error? 

Here's the short instructions as they 

1. Put the robot in Local mode
2. Call `ur_hardware_interface/play` with "ROS_external_control.urp" :)
3. Recovery steps:
   - Restart the driver OR call `ur_

fmauch, 2020-09-08:
The reason we're doing it that way is basically twofold:
 - As the socket connection isn't closed from the robot after the rejected call, you can still use calls that are possible in local mode
 - We would have to monitor for those mode changes and there is no information source for that that wouldn't have to be queried frequently. If you have an idea how to get around this, this is welcome.

felixvd, 2020-10-01:
Sorry for the late response. I implemented the `quit`/`connect` fallbacks in Python and C++, but it is not a very smooth experience. `loadProgram` can time out for no reason, and it is not clear which of the two services needs to be called when a call to a dashboard service has failed (connect? quit and connect?). [These scripts](https://gist.github.com/felixvd/8e11cff9ec533f34b89437bcfd44c64a) have been serving me well, so I uploaded them in case someone else would like to restart/reconnect the dashboard client comfortably.

I still don't really understand why `dashboard/connect` should not be called whenever the socket failed to connect. Is there an upside to the dashboard client/socket staying disconnected? It just seems like a defunct state to me that shouldn't require a manual reset. Feel free to close this if you prefer.

felixvd, 2020-10-07:
Related (dashboard usability): https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/262

urfeex, 2026-11-09:
Since ROS 1 is end of life, I will go ahead and close this.