UniversalRobots/Universal_Robots_ROS_Driver #260: Delay between the message "Sent program to robot" and "Robot ready to receive control commands"

sharpe-developer, 2020-09-09:
# Summary
I am experiencing an 8 to 10 second delay between the message "Sent program to robot" and "Robot ready to receive control commands". 

I am running an Ubuntu PC as the ROS machine. Real time patch is installed. The UR3 control box is connected directly via Ethernet to the Ubuntu PC (no switches, etc.). 

When I launch all goes well. I am using remote control mode and a python script that powers on the robot, loads the external control program, and runs it. This all works great but then I have to add a 15 second delay in my script before moving the robot since it is not ready to receive commands. As long as I add a delay before moving the robot then I can move the robot repeatedly without any issues. The delay seems excessive so I suspect something is going wrong.

My firewall is disabled as discussed in #8. I also tried updating to the newer external control urcap (v1.0.4). I looked at #1 and #237 but didn't see anything that stood out.

Questions:
1. Is this delay normal or is something causing a problem?
2. Is there a programmatic way to detect when the robot is ready to receive commands rather than using an arbitrary delay?

# Versions
 - Ubuntu 18.04.3 with Real Time patch installed
 - ROS Melodic
 - ROS Driver version: 0.0.3
 - Affected Robot Software Version(s): Polyscope 5.5 and 5.9
 - URCaps Software version(s): external control 1.0.2 (also tried 1.0.4)

# Issue details
Here is my log output. As you can see by the timestamps at the end it took approximately 8 seconds between sent program and ready to receive commands. The delay seems to be random in the range of 8 and 10 seconds.

```
... logging to /home/user/.ros/log/030633dc-f2c6-11ea-953e-98e7f4f4a84e/roslaunch-UR-ROS-5030.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://UR-ROS:44351/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 500
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'f...
 * /move_group/endeffector/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/manipulator/default_planner_config: RRTConnectkConfig...
 * /move_group/manipulator/longest_valid_segment_fraction: 0.01
 * /move_group/manipulator/planner_configs: ['SBLkConfigDefau...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
[... 373 lines cut ...]

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1599674290.348422130]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1599674290.348452654]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1599674316.691043347]: Robot requested program
[ INFO] [1599674316.691116911]: Sent program to robot
[ INFO] [1599674324.877954725]: Robot ready to receive control commands.
```

fmauch, 2020-09-09:
> * Is this delay normal or is something causing a problem?

No, this should not be the case. It is normal that it is not coming instantly, as the program is being compiled when started, but it shouldn't take that long.

> * Is there a programmatic way to detect when the robot is ready to receive commands rather than using an arbitrary delay?

You can listen to the [robot_program_running](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#robot_program_running-std_msgsbool) topic. But in that situation described I would consider this a workaround rather than a solution.

Are there any other URCaps involved? Does your program contain anything else than the ExternalControl? The program is fetched from the host machine during the start of the program. This is when you will see the `Sent program to robot` output. As soon as the program enters the ExternalControl program node.

sharpe-developer, 2020-09-09:
I performed a Wireshark capture of the interface and didn't see anything unusual. There is a steady stream of packets going back and forth during period between the message "Sent program to robot" and "Robot ready to receive control commands". I have attached the capture for reference if it is helpful.
[wireshark.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/5197657/wireshark.zip)

sharpe-developer, 2020-09-09:
There are other urcaps installed on the teach pendant but none are actively being used. I can try removing them to see if it makes a difference.

The program only contains the external control unit. There is nothing else in the program.

sharpe-developer, 2020-09-09:
I removed all the other urcaps and found it was the Robotiq urcap that was causing the problem. I had version Robotiq Grippers 1.6.0.4 installed. Removing the robotiq urcap brought the time between sent program and robot ready messages down to around 50 milliseconds.

I tried updating to Robotiq Grippers 1.8.4.4844 and the time between sent and ready was still milliseconds. It must have been something with the older version installed on the robot. 

However, I noticed a new problem. When the robot is rebooted and this setup is executed the dashboard server disconnects. If I re-launch the nodes on the ROS side (don't change anything on the robot) then everything works fine. So it is only the initial launch after restarting the robot that has an issue. If I remove the Robotiq urcap and restart the robot there are no problems on the initial launch. So the new Robotiq urcap does not have the delay but does not work perfectly either. Here is the log output when this error occurs>
```
You can start planning now!

[ INFO] [1599685906.093207380]: Robot mode is now POWER_ON
[ INFO] [1599685908.360545374]: Robot mode is now BOOTING
[ INFO] [1599685915.583168000]: Robot mode is now POWER_ON
[ INFO] [1599685916.116860271]: Robot mode is now IDLE
[ INFO] [1599685918.972324485]: Robot mode is now RUNNING
[ INFO] [1599685920.697553900]: Disconnecting from Dashboard server on 192.168.100.1:29999
[ERROR] [1599685920.697901565]: Exception thrown while processing service call: Did not receive answer from dashboard server in time. Disconnecting from dashboard server.(Configured timeout: 1 sec)
[ERROR] [1599685920.710130045]: Attempt to write on a non-connected socket
[ERROR] [1599685920.710239055]: Exception thrown while processing service call: Failed to send request to dashboard server. Are you connected to the Dashboard Server?
[ INFO] [1599685921.372080406]: Robot requested program
[ INFO] [1599685921.372174363]: Sent program to robot
[ INFO] [1599685921.566657241]: Robot ready to receive control commands.
```

I am curious. Why would having a urcap installed on the robot cause this when the urcap is not in use. Is having the urcap loaded on the robot enough to change how the robot operates?

gavanderhoorn, 2020-09-10:
> Is having the urcap loaded on the robot enough to change how the robot operates?

In general: no, but there are a lot of URCaps which, as part of their initialisation routine, either start or run or use additional processes and/or threads, which can do whatever.

As there is some isolation between URCaps, but not total (how could there be), there is certainly room for one URCap to influence the behaviour of others -- even if it doesn't do this intentionally.

> However, I noticed a new problem. 

To increase visibility of that problem, perhaps opening a new issue is in order?

sharpe-developer, 2020-09-10:
> To increase visibility of that problem, perhaps opening a new issue is in order?

Indeed. I will open a new issue.

closed 2020-09-10 by sharpe-developer