UniversalRobots/Universal_Robots_ROS_Driver #261: Control UR5 TCP pose: can Inverse kinematics include info from calibration file?

AnlageM, 2020-09-10:
Hi, I am new to ROS and I'd like to control UR5 TCP pose. I am using ROS noetic and I have an UR5 CB3. I managed to install the UR ROS driver and make it work, then I realized I should publish to the robot the joint positions. I suppose I need a kinematic solver as KDL to solve the kinematics, but I am missing something on how to integrate the calibration of the robot in this solver.

1) I installed the robot description in `fmauch_universal_robot`. How does `Universal_Robot_ROS_driver` merges the URDF of the typical UR5 robot with the info in the calibration file taken directly from my UR5 robot? Which is the script where I can find this passage?

The rest is maybe not directly related to the driver, but I would ask your opinion here because with your knowledge of the driver you could be faster than me in understanding whether it may work or not.
2) I need something fast enough in computing IK values, because I want to give the position to the robot as 'live' as possible. Is KDL or IKfast the best way to do that, is there already some specific code for UR or do I need to write my own solver?
3) Could [this](http://wiki.ros.org/Industrial/Tutorials/Create_a_Fast_IK_Solution#Generate_IK_Solver) be a good starting tutorial for my purpose?

Any suggestion is welcome and thanks for your attention and help.

# Impact
Calibrated IK

# Use Case and Setup
Teleoperation setup

fmauch, 2020-10-01:
Regarding your first question: Please see the README, specifically [this section](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#extract-calibration-information)

Regarding your second and third question: I've personally been using [our Cartesian controllers](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers) or [trac_ik](https://bitbucket.org/traclabs/trac_ik/src/master/) for Cartesian control.

We are currently working on implementing Cartesian control using the robot controller directly, but this isn't finished, yet.

AnlageM, 2020-10-02:
Hi fmauch, thanks you for your help. And thanks for your work with the driver and robot controller, I will follow for sure any update on the topic.

I've already read the readme and tried the URcalibration on my robot. What I cannot understand (this because I am new to ROS) and would like an hint on is: do either FK and IK algorithms of this driver exploit `my_robot_calibration.yaml` file? How do they account for it, and which is the script that does this within the `UR_ROS_driver`?

Could Cartesian controller or trac_ik you linked include the corrections of my_robot_calibration.yaml?
Any hint will be really welcome

fmauch, 2020-10-04:
As the urdf generated from ‘ur_description‘ will contain the values from the calibrated parameters file, any method using the urdf for ik (which to my knowledge includes all methods mentioned in this thread) will be using the calibrated robot information.

AnlageM, 2020-10-05:
Ok, I was on the default branch but I noticed that you are actually working to include calibration in URDF in melodic-devel-staging. So, this:
> the urdf generated from ‘ur_description‘ will contain the values from the calibrated parameters file

It happens only for melodc-devel-staging branch, am I right? Thanks for your help

fmauch, 2020-10-07:
> Ok, I was using the default branch of universal_robot, but I noticed that you @fmauch are actually working to include calibration in URDF in `melodic-devel-staging` branch. So:
> 
> 1. > the urdf generated from ‘ur_description‘ will contain the values from the calibrated parameters file
>    It happens only for melodic-devel-staging branch, does it?

For this driver we don't use this branch, yet, but the `calibration_devel` branch from my personal fork until `melodic-devel-staging` is at a state where we can to the switch. From both versions you will get a valid urdf, however.

> 2. To print the urdf created by `ur_description` in an .urdf file, I am launching `load_ur5.launch` using the `my_robot_calibration.yaml` as `kinematics_param` argument, and then use
>    `rosparam get -p /robot_description > myrobot.urdf`
>    Is this correct?

Yes, you can do that, but to generate a urdf file from an xacro file, you can use the `xacro` command that is used inside the `load_urX.launch` directly. See [this wiki entry](http://wiki.ros.org/urdf/Tutorials/Using%20Xacro%20to%20Clean%20Up%20a%20URDF%20File) as a simple example.

fmauch, 2020-11-16:
Closing due to inactivity. Feel free to comment or reopen if anything new comes up.

closed 2020-11-16 by fmauch