UniversalRobots/Universal_Robots_ROS_Driver #262: Dashboard server disconnects on first execution after robot restart

sharpe-developer, 2020-09-10:
# Summary

Dashboard server disconnects on first run after robot restart with the Robotiq Gripper urcap installed on the teach pendant.

# Versions

 - Ubuntu 18.04.3 with Real Time patch installed
 - ROS Melodic
 - ROS Driver version: 0.0.3
 - Affected Robot Software Version(s):  5.9.1.1031110
 - URCaps Software version(s): external control 1.0.4, rs485 1.0.0, Robotiq grippers 1.8.4.4844

# Impact
Dashboard server crashes on first run after robot restart when Robotiq Gripper urcap installed

# Issue details
The first time a sequence of dashboard commands are executed after a robot restart then the dashboard server disconnects/crashes. After the first error occurs the driver can be relaunched and the dashboard commands can be repeated multiple times without any problems. However if the robot is restarted/rebooted then the dashboard server will disconnect the first time the steps are executed.

The robot is directly connected to the ROS PC via Ethernet (i.e. no switches, routers, etc.)

## Steps to Reproduce

On the teach pendant

* Restart robot
* Put robot in remote control mode

Open terminal and launch robot driver

```
roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:="192.168.100.1" reverse_port:="29999"
```

Power the robot via dashboard

```
$ rosservice call /ur_hardware_interface/dashboard/power_on 
success: True
message: "Powering on"
```

Wait for robot mode = IDLE

Release the brake

```
$ rosservice call /ur_hardware_interface/dashboard/brake_release 
success: True
message: "Brake releasing"
```

Wait for robot mode = RUNNING

Load the external control program via dashboard

```
$ rosservice call /ur_hardware_interface/dashboard/load_program "ExternalControl.urp"
answer: "Loading program: /programs/ExternalControl.urp, /programs/default.installation"
success: True
```

Set speed slider

```
$ rosservice call /ur_hardware_interface/set_speed_slider 0.5
success: True
```

Play the loaded program
Dashboard server disconnects
[... 284 lines cut ...]
[INFO] [1599759659.619706]: Loading controller: pos_joint_traj_controller
[ INFO] [1599759659.620889572]: Service available.
[INFO] [1599759659.665229]: Loading controller: joint_group_vel_controller
[INFO] [1599759659.675259]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[ INFO] [1599759659.732085263]: Robot's safety mode is now NORMAL
[ INFO] [1599759659.733309251]: Robot mode is now POWER_OFF
[INFO] [1599759659.909708]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1599759659.912711]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1599759659.915378]: Loading controller: joint_state_controller
[INFO] [1599759659.923278]: Loading controller: scaled_pos_joint_traj_controller
[INFO] [1599759659.959220]: Loading controller: speed_scaling_state_controller
[INFO] [1599759659.966954]: Loading controller: force_torque_sensor_controller
[INFO] [1599759659.975131]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1599759659.979061]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[ INFO] [1599759674.033885929]: Robot mode is now POWER_ON
[ INFO] [1599759676.297903724]: Robot mode is now BOOTING
[ INFO] [1599759682.035534410]: Robot mode is now POWER_ON
[ INFO] [1599759682.635326742]: Robot mode is now IDLE
[ INFO] [1599759702.553589570]: Robot mode is now RUNNING
[ INFO] [1599759878.214629918]: Disconnecting from Dashboard server on 192.168.100.1:29999
[ERROR] [1599759878.214938824]: Exception thrown while processing service call: Did not receive answer from dashboard server in time. Disconnecting from dashboard server.(Configured timeout: 1 sec)
[ INFO] [1599759878.953081252]: Robot requested program
[ INFO] [1599759878.953162364]: Sent program to robot
[ INFO] [1599759879.142996550]: Robot ready to receive control commands.
```

fmauch, 2020-09-11:
> roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:="192.168.100.1" reverse_port:="29999"

Is that on purpose that you use the same port number on the ROS machine for the reverse port as used for the dashboard server on the robot? It probably won't hurt, I am just curious...

Do you see any logs on the teach pendant that could relate to that? If you could extract and post the generated program as explained [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/74#issuecomment-571949829), that could shed some light onto this.

sharpe-developer, 2020-09-11:
> Is that on purpose that you use the same port number on the ROS machine for the reverse port as used for the dashboard server on the robot? It probably won't hurt, I am just curious...

It is on purpose but that's likely a misunderstanding on my part. I was getting this error when in manual mode
```
"Command is not allowed due to safety reasons, please switch robot to Remote Control mode and reconnect to port 29999"
```
I assumed that meant the reverse port needed to be 29999 while in remote control mode. 
- Is that incorrect? 
- Can I use the default port of 50001 when in remote control mode? 
- When it states reconnect to port 29999 which port/interface is the error message referring to? 
- Is there any doc/diagram that shows the various connections that are made between the robot and ROS PC to help understand/visualize the communication setup?

> Do you see any logs on the teach pendant that could relate to that? If you could extract and post the generated program as explained here, that could shed some light onto this.

I will look at the logs and extract the program when I have access to the robot next week.

fmauch, 2020-09-14:
> It is on purpose but that's likely a misunderstanding on my part. I was getting this error when in manual mode
>
> ```
> "Command is not allowed due to safety reasons, please switch robot to Remote [...]

That message is coming from the dashboard server itself and only says that you should reconnect the dashboard client. This is always necessary when switching between local and remote mode when there is a dashboard client connected. See the [quit](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#dashboardquit-ur_dashboard_msgsgetloadedprogram) and[connect](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#dashboardquit-ur_dashboard_msgsgetloadedprogram) services to do a reconnect.

> * Is that incorrect?

See my explanation above

> * Can I use the default reverse port of 50001 when in remote control mode?

Yes, absolutely.

> * When it states `reconnect to port 29999` which port/interface is the error message referring to?

See above.

> * Is there any doc/diagram that shows the various connections that are made between the robot and ROS PC to help understand/visualize the communication setup?

We've just recently published a standalone library with a more in-depth [communication structure documentation](https://github.com/UniversalRobots/Universal_Robots_Client_Library#architecture).

sharpe-developer, 2020-09-15:
> Do you see any logs on the teach pendant that could relate to that? If you could extract and post the generated program as explained here, that could shed some light onto this.

I did not see anything unusual when I scanned to logs on the teach pendant UI. Here are the files you requested:
[files.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/5228607/files.zip)

fmauch, 2020-09-15:
I cannot see any trace of the robotique URCap inside `ExternalControl.script`, however, you state that the problem only occurs, when that URCap is installed.

Does the problem also occur, if you create a simple program (e.g. only containing a `wait` command) and load and play this?

sharpe-developer, 2020-09-21:
> Does the problem also occur, if you create a simple program (e.g. only containing a wait command) and load and play this?

No. I created a program with a single wait command. When I repeat the steps to reproduce (only changing the loaded program name) the problem does not occur.

fmauch, 2020-09-22:
Hm, ok... As stated above I don't see any interaction between the driver and the robotiq URCap, so I currently cannot provide any hints where the problem could lie,

sharpe-developer, 2020-09-22:
Seems like this might be an issue for the Robotiq URCap creator. Shall we close this issue since it seems nothing unusual on the UR robot driver side?

fmauch, 2020-09-23:
As this only occurs when you have the external control program node in your program it does not seem to be completely unrelated.

jappoz, 2022-09-05:
Hello, 
we are having a similar problem without using the Robotiq Urcap.
Are there any news on that?

floxdeveloper, 2022-09-22:
I have got the same problem as well.  When trying to experiment with the timeout mentioned in the error message, I recognized that the parameter received_timeout (https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#dashboardreceive_timeout-required) is hard coded in the dashboard_client_ros.cpp.
Another observation: The robot runs the program it should play, despite the error message. But afterwards I can't send any new Play-Commands that will be executed to the robot.

MilanPadhiyar027, 2022-09-29:
Hi All,

I am also facing a similar issue mentioned above even without using the Robotiq URCap program node. If you just use the external control program node then also the same errors occurs for the first time execution. @fmauch can you please share if you have any updates here? 

Thanks

floxdeveloper, 2022-09-29:
I got it working after I removed the [ExternalControl-URCap-Plugin](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap).

MilanPadhiyar027, 2022-09-29:
> I got it working after I removed the [ExternalControl-URCap-Plugin](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap).

Can you please explain more about this? How can use the external control without using the mentioned plugin?

floxdeveloper, 2022-09-29:
I am not completely sure, if my setup is correct now. But I just got under Settings > URCaps > ExternalControl > -

There might be some remaining configurations of the previously installed plugin.

I am using the Universal_Robots_ROS_Driver to control my robot, when it's in "Remote control", I have never used the "External Control"-Node of the ExternalControl-URCap.

MilanPadhiyar027, 2022-09-29:
> I am not completely sure, if my setup is correct now. But I just got under Settings > URCaps > ExternalControl > -
>
> There might be some remaining configurations of the previously installed plugin. [...]

which program are you configuring into the program node of the ROS driver?

floxdeveloper, 2022-09-29:
What do you mean by program node of the ROS driver? 

I am connecting my UR5 and my laptop via ethernet, turn the UR5 in remote mode and start ROS with the following command:
`roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.12.1   kinematics_config:=/home/user/ur5e.yaml` 

Afterwards I use commands like:
`rosservice call /ur_hardware_interface/dashboard/brake_release`
`rosservice call /ur_hardware_interface/dashboard/load_programm home_position.urp`
`rosservice call /ur_hardware_interface/dashboard/play`

And `home_position.urp` contains a `MoveL`-program node with a position.

MilanPadhiyar027, 2022-09-30:
Can you please share a screenshot of your home_position.urp program of a teach pendant? I mean in the program node there should be one node for the external program to control the robot via the ROS interface.

fmauch, 2022-09-30:
(Maybe) related to this topic there's currently https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/118.

@MilanPadhiyar027 When having the robot in remote_control mode, you can also start the driver in [headless_mode](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/README.md#headless-mode). This will not require the ExternalControl URCap running on the robot.

If you could try whether https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/118 helps out, that would be very much appreciated.

elvintoh82, 2022-10-27:
> I had a similar error "Did not receive answer from dashboard server in time. Disconnecting from dashboard server.(Configured timeout: 1 sec)".

I subsequently found line 360 in 'dashboard_client_ros.cpp' and changed it from
`  tv.tv_sec = nh_.param("receive_timeout", 1);
`
to

`  tv.tv_sec = nh_.param("receive_timeout", 2);
`
I then did a catkin clean followed by a catkin build. But alas, the same error recurred when I called dashboard/play again. But I would have expected it to show ...(Configured timeout: 2 sec). 
But instead it showed the exact same message (Configured timeout: 1 sec).
Indicating that my change from 1 to 2 second was not implemented actually. 

By way of tracing, can someone tell me where exactly (in which file/line) was the error message "(Configured timeout: XX sec)" printed out from please? For the life of me, I just can't seem to find/trace where that was printed out from.

Any advice?

urfeex, 2026-11-09:
Since ROS 1 is end of life, I will go ahead and close this.

closed 2026-11-09 by urfeex