UniversalRobots/Universal_Robots_ROS_Driver #263: How to set TCP force control mode？

Csp0920, 2020-09-11:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*

By using polyscope，I can set a force that causes the robot to move until another force is balanced with that force.
Now I am not going to use polyscope. But I don't know if there is an api that can realize the same result.
I see the URScript's api, but I am not clear about the communication between PC(ROS) and real UR3e robot.

I have seen a problem published in the issue of Universal_Robots_ROS_Driver. But I find that publishing a urscript to ur_hardware_interface/script_command will never get the feedback. 

So how can I sovle this problem? 

Looking forward for your reply. Thank you very much. 

# Versions
 - ROS Driver version: 16.04kinetic
 - Affected Robot Software Version(s):UR3e
 - Affected Robot Hardware Version(s):UR3e
 - Robot Serial Number:
 - UR+ product(s) installed:no
 - URCaps Software version(s): 1.0.1

fmauch, 2020-09-11:
This is currently not supported. You can, however, send arbitrary script code for execution to the robot. In this case, you won't be able to use the ROS controllers.

Csp0920, 2020-09-16:
> 当前不支持此功能。但是，您可以将任意脚本代码发送给机器人。在这种情况下，您将无法使用ROS控制器。

Thanks for your reply. 
I see the question#219 that you said the force_torque_sensor_controller will publish the wrench on /wrench. I carefully check the code, but I haven't found the publisher that can realize this function. Only a code in the ur_driver of UR package:
pub_wrench = rospy.Publisher('wrench', WrenchStamped, queue_size=1)
So I want to know that if the ur_driver has been abandoned? And how can I publish the wrench on other topic?

I use the package you mentioned in #219(https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers), it subscribe a topic named "ft_sensor_wrench", I want to publish the wrench on /ft_sensor_wrench. 

Thank for your patient.

Csp0920, 2020-09-18:
I have solved this question, it is mainly because I don't understand the code rules.

closed 2020-09-18 by Csp0920