UniversalRobots/Universal_Robots_ROS_Driver #267: Jerky motion even when using vel_traj_controller of ur_robot_driver

blackhorsewu, 2020-09-16:
*Introduction to the issue*

# Versions
 - ROS Driver version: ur_robot_driver
 - Affected Robot Software Version(s): URSoftware 3.10
 - Affected Robot Hardware Version(s):
 - Robot Serial Number: UR5
 - UR+ product(s) installed:
 - URCaps Software version(s):

# Issue details

When I was using `ur_modern_driver` on ur3 last year, the `ros vel_based_pos_traj_controller`, the manipulator can move smoothly. Now, I am trying the `vel_traj_controller` of the `ur_robot_driver` on ur5, the movement of the manipulator is very jerky!

I just noticed that `vel_traj_controller` is different from `vel_based_pos_traj_controller`. I suppose I should have used `vel_based_pos_traj_controller` if it ever exist! If it is not available yet, when would it be available at all?

Is there any thing that can be done to smooth the motion for the time being?

Thank you for your help.

gavanderhoorn, 2020-09-16:
> Now, I am trying the `vel_traj_controller`

Please tell us where you find this controller defined.

I cannot find it in the files in [ur_robot_driver/config](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/406a3308d8d45906d83acd7fa4263cd6476223f3/ur_robot_driver/config).

blackhorsewu, 2020-09-16:
In the `ur_robot_driver/config/ur5_controllers.yaml` file, there are `scaled_vel_traj_controller` and `vel_traj_controller sections`.

gavanderhoorn, 2020-09-16:
Please link to specific line number in the file.

Afaict, that file contains these entries for velocity related controllers:

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/406a3308d8d45906d83acd7fa4263cd6476223f3/ur_robot_driver/config/ur5_controllers.yaml#L64-L131

fmauch, 2020-09-16:
@blackhorsewu I suppose you did not pull the latest version as those got renamed in 7ac5b4d235cb3cfd95f1c2ebe052c776e2e3263a 2 months ago.

Furthermore, I am assuming you are referring to what we now call `vel_joint_traj_controller`.

blackhorsewu, 2020-09-16:
@fmauch and @gavanderhoorn No, I did not pull the latest version. I was just checking the dates of the files. The last modified dates of my files are 6 June and the dates on the github files are 28 days ago. Anyway the contents of these sections are the same, as far as I can tell.

blackhorsewu, 2020-09-16:
Actually, when I use the `vel_traj_controller` it is already better that `scaled_vel_traj_controller`.

gavanderhoorn, 2020-09-16:
To avoid confusion: could you please copy-paste the contents of your `ur5_controllers.yaml` here in a comment (be sure to paste it into a [markdown code block](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks)).

I find it very hard to reason about what you are doing, as you refer to names which are no longer used here.

fmauch, 2020-09-16:
And as you can see in

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/406a3308d8d45906d83acd7fa4263cd6476223f3/ur_robot_driver/config/ur5_controllers.yaml#L109

the pid values for the controller aren't optimized in any way. 

If you like, you can try adapting those values to the ones from the ur_modern_driver

https://github.com/ros-industrial/ur_modern_driver/blob/kinetic-devel/config/ur5_controllers.yaml#L77-L83

If that helps, please let us know.

blackhorsewu, 2020-09-16:
@gavanderhoorn , the following is the content of my `ur5_controllers.yaml`
```
# Settings for ros_control control loop
hardware_control_loop:
   loop_hz: &loop_hz 125

# Settings for ros_control hardware interface
hardware_interface:
   joints: &robot_joints
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint

# Publish all joint states ----------------------------------
joint_state_controller:
   type:         joint_state_controller/JointStateController
   publish_rate: *loop_hz

# Publish wrench ----------------------------------
force_torque_sensor_controller:
   type:         force_torque_sensor_controller/ForceTorqueSensorController
   publish_rate: *loop_hz

# Publish speed_scaling factor
speed_scaling_state_controller:
   type:         ur_controllers/SpeedScalingStateController
   publish_rate: *loop_hz

# Joint Trajectory Controller - position based -------------------------------
# For detailed explanations of parameter see http://wiki.ros.org/joint_trajectory_controller
scaled_pos_traj_controller:
   type: position_controllers/ScaledJointTrajectoryController
   joints: *robot_joints
   constraints:
      goal_time: 0.6
      stopped_velocity_tolerance: 0.05
      shoulder_pan_joint: {trajectory: 0.2, goal: 0.1}
      shoulder_lift_joint: {trajectory: 0.2, goal: 0.1}
      elbow_joint: {trajectory: 0.2, goal: 0.1}
      wrist_1_joint: {trajectory: 0.2, goal: 0.1}
      wrist_2_joint: {trajectory: 0.2, goal: 0.1}
      wrist_3_joint: {trajectory: 0.2, goal: 0.1}
   stop_trajectory_duration: 0.5
   state_publish_rate: *loop_hz
   action_monitor_rate: 10

pos_traj_controller:
   type: position_controllers/JointTrajectoryController
   joints: *robot_joints
   constraints:
      goal_time: 0.6
      stopped_velocity_tolerance: 0.05
      shoulder_pan_joint: {trajectory: 0.2, goal: 0.1}
      shoulder_lift_joint: {trajectory: 0.2, goal: 0.1}
      elbow_joint: {trajectory: 0.2, goal: 0.1}
      wrist_1_joint: {trajectory: 0.2, goal: 0.1}
      wrist_2_joint: {trajectory: 0.2, goal: 0.1}
      wrist_3_joint: {trajectory: 0.2, goal: 0.1}
   stop_trajectory_duration: 0.5
   state_publish_rate: *loop_hz
   action_monitor_rate: 10

scaled_vel_traj_controller:
   type: velocity_controllers/ScaledJointTrajectoryController
   joints: *robot_joints
   constraints:
      goal_time: 0.6
      stopped_velocity_tolerance: 0.05
      shoulder_pan_joint: {trajectory: 0.1, goal: 0.1}
      shoulder_lift_joint: {trajectory: 0.1, goal: 0.1}
      elbow_joint: {trajectory: 0.1, goal: 0.1}
      wrist_1_joint: {trajectory: 0.1, goal: 0.1}
      wrist_2_joint: {trajectory: 0.1, goal: 0.1}
      wrist_3_joint: {trajectory: 0.1, goal: 0.1}
   stop_trajectory_duration: 0.5
   state_publish_rate:  *loop_hz
   action_monitor_rate: 10
   gains:
      #!!These values have not been optimized!!
      shoulder_pan_joint:  {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      shoulder_lift_joint: {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      elbow_joint:         {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      wrist_1_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      wrist_2_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      wrist_3_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}

   # Use a feedforward term to reduce the size of PID gains
   velocity_ff:
      shoulder_pan_joint: 1.0
      shoulder_lift_joint: 1.0
      elbow_joint: 1.0
      wrist_1_joint: 1.0
      wrist_2_joint: 1.0
      wrist_3_joint: 1.0

   # state_publish_rate:  50 # Defaults to 50
   # action_monitor_rate: 20 # Defaults to 20
   #stop_trajectory_duration: 0 # Defaults to 0.0

vel_traj_controller:
   type: velocity_controllers/JointTrajectoryController
   joints: *robot_joints
   constraints:
      goal_time: 0.6
      stopped_velocity_tolerance: 0.05
      shoulder_pan_joint: {trajectory: 0.1, goal: 0.1}
      shoulder_lift_joint: {trajectory: 0.1, goal: 0.1}
      elbow_joint: {trajectory: 0.1, goal: 0.1}
      wrist_1_joint: {trajectory: 0.1, goal: 0.1}
      wrist_2_joint: {trajectory: 0.1, goal: 0.1}
      wrist_3_joint: {trajectory: 0.1, goal: 0.1}
   stop_trajectory_duration: 0.5
   state_publish_rate:  *loop_hz
   action_monitor_rate: 10
   gains:
      #!!These values have not been optimized!!
      shoulder_pan_joint:  {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      shoulder_lift_joint: {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      elbow_joint:         {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      wrist_1_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      wrist_2_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      wrist_3_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}

   # Use a feedforward term to reduce the size of PID gains
   velocity_ff:
      shoulder_pan_joint: 1.0
      shoulder_lift_joint: 1.0
      elbow_joint: 1.0
      wrist_1_joint: 1.0
      wrist_2_joint: 1.0
      wrist_3_joint: 1.0

   # state_publish_rate:  50 # Defaults to 50
   # action_monitor_rate: 20 # Defaults to 20
   #stop_trajectory_duration: 0 # Defaults to 0.0

# Pass an array of joint velocities directly to the joints
joint_group_vel_controller:
   type: velocity_controllers/JointGroupVelocityController
   joints: *robot_joints

robot_status_controller:
   type: industrial_robot_status_controller/IndustrialRobotStatusController
   handle_name: industrial_robot_status_handle
   publish_rate: 10

controller_list:
  - name: ""
#    action_ns: joint_trajectory_action
    action_ns: scaled_vel_traj_controller/follow_joint_trajectory
    type: FollowJointTrajectory
    joints: [shoulder_pan_joint, shoulder_lift_joint, elbow_joint, wrist_1_joint, wrist_2_joint, wrist_3_joint] 
```

blackhorsewu, 2020-09-16:
@fmauch , I will try comparing the parameters first and then try to see if those from the `ur_modern_driver` helps and report back here.

Thanks to both @fmauch and @gavanderhoorn for the help.

blackhorsewu, 2020-09-16:
I have just copied the parameters from the ur_modern_driver and the result is not much different! 
Another thing that I am thinking of: could it be because I have not installed the realtime kernel for the PC that affects the behaviour? For the realtime kernel is compatible with CUDA!

gavanderhoorn, 2020-09-16:
Perhaps it would be good if you first describe what it is you're actually trying to do.

Do you want to execute complete trajectories, or perform real-time control using velocity setpoints?

blackhorsewu, 2020-09-16:
I am still trying to use Descartes to generate a complete trajectory and then send it to the trajectory follower to excute it. I am not trying to interfere the trajectory in real-time yet.

gavanderhoorn, 2020-09-16:
Do those trajectories have proper `time_from_start` and velocities and accelerations?

fmauch, 2020-09-16:
In that case I would recommend to use the `scaled_pos_joint_traj_controller`. That should provide the smoothest trajectory execution.

blackhorsewu, 2020-09-16:
@gavanderhoorn That is what I am not sure! When I generate the trajectory for an arc, it complaint that the time_from_start are not strictly increasing. When I printed them out, some of them are the same. I completely do not know what the velocities and accelerations putted in by Descartes.
Or else is there a way that I can work them out myself?

blackhorsewu, 2020-09-16:
Do I need a trajectory filter to make sure the timing, velocity and acceleration make sense?

DariusStu, 2020-09-21:
Do you see the same jerkiness when using the joint_group_vel_controller?

blackhorsewu, 2020-09-22:
Thank you very much for your email.

 

I cannot use this controller for it says it need an array of joint velocities directly for the joints, that I do not have.

 

Regards,

Victor Wu.

fmauch, 2020-09-22:
> I cannot use this controller for it says it need an array of joint velocities directly for the joints, that I do not have.

But you could give it a try e.g. using [this python script](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/231#issuecomment-688415091).

However, I don't think that this will bring any benefits to solving this issue, as you clearly stated that you have non-optimal trajectory timings. You can test sending a simple trajectory where you know the timings are correct. If this is behaving correctly, then your problem lies in generating your trajectories which is if I understood correctly a matter with descartes.

gavanderhoorn, 2020-09-22:
It's very likely that a trajectory without proper velocity and accelerations is the cause, as @fmauch writes.

As to why `ur_modern_driver` did not suffer from this, I'm not sure, but perhaps the gains used with the underlying URScript functions were configured differently.

blackhorsewu, 2020-09-22:
As the trajectory point is generated by a call to the `descartes_utilities::toRosJoinPoints`. There, the `ros_pt.velocities` are set to 0.0, `ros_pt.accelerations` are set to 0.0 and `ros_pt.effort` is also set to 0.0. I am trusting that the `ur_robot_driver` will fill in the proper and optimal values for them.
Would you suggest a way to get or calculate suitable values for them?

gavanderhoorn, 2020-09-22:
> I am trusting that the `ur_robot_driver` will fill in the proper and optimal values for them.

No, it won't.

> Would you suggest a way to get or calculate suitable values for them?

You could take a look at the available time parameterisation plugins MoveIt offers: [Time Parameterization](https://ros-planning.github.io/moveit_tutorials/doc/time_parameterization/time_parameterization_tutorial.html).

blackhorsewu, 2020-09-22:
@gavanderhoorn thank you very much for spotting the cause of my problem and explained the reason. I will definitely try to learn the `Time Parameterization`. Should I now close this issue?

fmauch, 2020-09-22:
If you feel that this issue is resolved, you can go ahead and close it. However, the timings currently are only the expected cause of your problems. If I was in your position I'd try to send a trajectory to the driver that uses valid timestamps and verify that the execution is as expected. If this is the case I would close this issue.

blackhorsewu, 2020-09-22:
@fmauch thank you very much. The `time_from_start` sent, for a straight line, seemed sensible and the movement was also reasonable except that it was jerky, as @gavanderhoorn said, the end effector was stopping and starting at every waypoint.

blackhorsewu, 2020-09-22:
So, I would have to learn how to work out the velocities and accelerations of evey joint at every waypoint so that the manipulator will move smoothly along the trajectory.

fmauch, 2020-10-01:
As related to this driver - can this issue be closed?

closed 2020-10-12 by blackhorsewu

DariusStu, 2021-01-12:
Is there a generic troubleshooting list for experiencing jerky movements on the hardware when using the joint_group_vel_controller? Are there common parameters or settings to change to generate smoother motions? I’ve read several of the open & closed issues, and they don’t provided the necessary resolution I’d hope for. @fmauch @gavanderhoorn @blackhorsewu

fmauch, 2021-01-14:
@DariusStu When using the `joint_group_vel_controller` robot motion reflects the inputs almost instantly. Depending on your use-case and setup it's hard to tell where this comes from. Maybe open a new issue describing your setup and the issues you are facing. Please be aware that this controller takes the value read from the input topic and sends it directly to the robot, where it is fed into a `speedj` command. Therefore, if the input values are jerk-free, motion should also be jerk-free. However, topic communication does not necessarily happen anywhere close to real-time requirements.