UniversalRobots/Universal_Robots_ROS_Driver #269: I have a problem on force_torque_sensor_controller, and its topic /wrench

Csp0920, 2020-09-19:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*
Recently, I used force_torque_sensor_controller, and subscribed the topic /wrench. I could successfuly get the imformation, but why the force was no 0 when the robot arm was static?
After moving robot(ur3e) serval times, I put the robot on the same position, but the imformation of  force was different.

Csp0920, 2020-09-19:
I forget to spawn the robot on external control. .....

Csp0920, 2020-09-21:
Hello, now I have an another question:
When I successfully get the data of /wrench, I find that the "force" data are becoming bigger and bigger.
I have no idea on what causes this.

Csp0920, 2020-09-21:
And the "torque" data are correct I personally believe.

fmauch, 2020-09-21:
The driver is only forwarding the readings from the robot and rotates them to be in the correct frame. Any drift that occurs in the data is independent of this driver.

davideferrari95, 2020-10-29:
> The driver is only forwarding the readings from the robot and rotates them to be in the correct frame. Any drift that occurs in the data is independent of this driver.

I have the same problem with the UR10e built-in  FT-Sensor. The readings from the sensor getting bigger in Fx, Fy and Fz as shown in next plots. They are obtained by leaving the robot in a custom position for about 20 minutes, connected with ROS, and saving all data from topic "/wrench". The results clearly show an increasing error in all force direction.

Is it a sensor defect or a problem with the UR control software ? Is there a solution?

Thanks, Davide

Fx:
[image]
Fy:
[image]
Fz:
[image]

Csp0920, 2020-10-29:
> I have the same problem with the UR10e built-in FT-Sensor. The readings from the sensor getting bigger in Fx, Fy and Fz as shown in next plots. They are obtained by leaving the robot in a custom [...]

Recently, I sent the UR3e robot to the agent's Testing Center for inspection. 
Unfortunately, there is no new progress.
I can't see your picture. Is the force growing bigger without limit?

davideferrari95, 2020-10-29:
> > > The driver is only forwarding the readings from the robot and rotates them to be in the correct frame. Any drift that occurs in the data is independent of this driver.
> > 
> > 
> > I have the same problem with the UR10e built-in FT-Sensor. The readings from the sensor getting bigger in Fx, Fy and Fz as shown in next plots. They are obtained by leaving the robot in a custom position for about 20 minutes, connected with ROS, and saving all data from topic "/wrench". The results clearly show an increasing error in all force direction.
> > Is it a sensor defect or a problem with the UR control software ? Is there a solution?
> > Thanks, Davide
> 
> Recently, I sent the UR3e robot to the agent's Testing Center for inspection.
> Unfortunately, there is no new progress.
> I can't see your picture. Is the force growing bigger without limit?

Exactly, force is growing bigger in -X, -Y, +Z with different slope.
From the tests carried out it would seem to grow without a limit. I have new tests scheduled tomorrow to compare the results.

I've uploaded the images on imgur, I hope you can see them there.

Fx: https://imgur.com/4x05SqX
Fy: https://imgur.com/hMcP8Cb
Fz: https://imgur.com/XK7Qple

gavanderhoorn, 2020-10-29:
Is this before or after you've 'tared' the sensor?

davideferrari95, 2020-10-29:
> Is this before or after you've 'tared' the sensor?

I call the service "zero_ftsensor" each time my node start and I added even the "zero_ftsensor()" function in TP when the ROS comunication program is launched.

Every time I call the zeroing service the sensors set its values to zero, then it starts again to sense growing forces.

Csp0920, 2020-10-29:
I also used the code "zero_ftsensor", but the force still grow without limit. 
Maybe something wrong in the hardware?

imishani, 2021-01-13:
It happens to me to using ft300. The force values keeps growing over time and when I load the sensor with force and then release it to the same position i get deviation.
Have no idea what to do. if you found a solution i'll be more than happy to hear.

imishani, 2021-01-13:
By the way, I have plugged the sensor to the computer and using ROS.

davideferrari95, 2021-01-13:
> It happens to me to using ft300. The force values keeps growing over time and when I load the sensor with force and then release it to the same position i get deviation.
> Have no idea what to do. if you found a solution i'll be more than happy to hear.

The only solution I found is to call the "zero_ftsensor" service every time I start my codes.

imishani, 2021-01-13:
> The only solution I found is to call the "zero_ftsensor" service every time I start my codes.

Its ridiculous. If i need to measure continuously i cant zero the sensor.

gavanderhoorn, 2021-01-13:
As @fmauch already wrote, the driver does not do anything with the measurements reported by the sensor. It would seem more appropriate to report problems with the sensor on the UR support forum.

@imishani wrote:
> It happens to me to using ft300. [..] I have plugged the sensor to the computer and using ROS.

this issue discusses problems with the built-in FT sensor.

Your setup appears different, and this issue seems not like it would be a good place to discuss your problems.

DariusStu, 2021-05-24:
Has there been any update with this issue? Did UR respond saying what could be the issue using the internal FT-sensor on the manipulators? @davideferrari95

davideferrari95, 2021-05-24:
No news, we got around this by scheduling short experiment sessions and calibrating the sensor each time.

DariusStu, 2021-05-24:
Calibrating the sensor on each joint? Also, did you happen to discover why the internal FT-sensors were off for the manipulator? I've read through the user manuals by Universal Robotics, and can't seem to find how the FT values are computed. Any advice here? @davideferrari95 or @gavanderhoorn  Thanks!

davideferrari95, 2021-05-24:
@DariusStu No, we calibrate only the FT-Sensor using the zero ft sensor service in headless mode.
We think the problem is related to some thermal throttling but we have not investigated further.

gavanderhoorn, 2021-05-24:
> Calibrating the sensor on each joint?

I'm not a UR employee, but afaik, unless something has recently changed, UR robots do not have torque sensors in each joint. The efforts reported by the driver are actually the currents reported by the controller.

urrsk, 2021-05-25:
The FT sensor in the E-serie is a relative sensor that are temperature dependent. Over a short time, the drift is very little, on a robot that have reach its stable temperature. Therefore, it is beneficial to tara the sensor just before using the signal. This can be done by the ```zero_ftsensor()``` URScript command or by subtraction an offset from the ROS side. This is how it is done using the Force Mode node in Polyscope.

In the other hand, the internal FT sensor is very fast and have a high mechanical bandwidth. This is utilize used in the Polyscope's "move until contact" functionality or the ```tool_contact()``` URScript function.

stefanscherzinger, 2021-09-01:
Please feel free to re-open with additional questions or insights.

closed 2021-09-01 by stefanscherzinger