UniversalRobots/Universal_Robots_ROS_Driver #271: Other ros controllers stopped?

dniewinski, 2020-09-24:
When launching the driver, all other controllers in ros controller manager are stopped and removed.

# Versions
 - ROS Driver version: 406a3308d8d45906d83acd7fa4263cd6476223f3
 - Affected Robot Software Version(s): 3.13
 - Affected Robot Hardware Version(s): UR5
 - Robot Serial Number: No idea
 - UR+ product(s) installed: None
 - URCaps Software version(s): 1.0.3

# Impact
This breaks the rest of the platform.  All other ros controllers stop working 

# Issue details
## Use Case and Setup
- Mobile platform is running, and has controllers
- Arm is connected to the same ros computer that is running the mobile platform
- Arm driver is started

## Project status at point of discovered
- UR Driver works individually
- Mobile platform works individually

## Steps to Reproduce
- Have ros controllers running
- Launch UR driver

## Expected Behavior
That the currently running controllers would not be affected and the UR controllers would just be spawned

## Actual Behavior
### Verify Platform Controllers are Running

    $ rosrun controller_manager controller_manager list
    'husky_joint_publisher' - 'hardware_interface::JointStateInterface' ( running )
    'husky_velocity_controller' - 'hardware_interface::VelocityJointInterface' ( running )

### Launch UR Driver
Get errors about unable to load controllers

    [ERROR] [1600953420.394999]: Failed to load pos_joint_traj_controller
    [INFO] [1600953420.397104]: Controller Spawner: Loaded controllers: 
    [ERROR] [1600953420.399550]: Failed to load scaled_pos_joint_traj_controller
    [INFO] [1600953420.401473]: Loading controller: speed_scaling_state_controller
    [ERROR] [1600953421.410755]: Failed to load speed_scaling_state_controller
    [INFO] [1600953421.412729]: Loading controller: force_torque_sensor_controller
    [ERROR] [1600953422.421385]: Failed to load force_torque_sensor_controller
    [INFO] [1600953422.423497]: Loading controller: robot_status_controller
    [ERROR] [1600953423.432280]: Failed to load robot_status_controller

Previously running controllers fail and are removed

    [ERROR] [1600953426.842046565]: Could not start controller with name 'husky_joint_publisher' because no controller with this name exists

    $ rosrun controller_manager controller_manager list
    No controllers are loaded in mechanism control

### Re Launch UR Driver
Everything UR starts as expected

    $ rosrun controller_manager controller_manager list
    'joint_state_controller' - 'hardware_interface::JointStateInterface' ( running )
    'pos_joint_traj_controller' - 'hardware_interface::PositionJointInterface' ( initialized )
    'scaled_pos_joint_traj_controller' - 'ur_controllers::ScaledPositionJointInterface' ( running )
    'speed_scaling_state_controller' - 'ur_controllers::SpeedScalingInterface' ( running )
    'force_torque_sensor_controller' - 'hardware_interface::ForceTorqueSensorInterface' ( running )
    'robot_status_controller' - 'industrial_robot_status_interface::IndustrialRobotStatusInterface' ( running )

dniewinski, 2020-09-24:
Some possibly related questions:

- Are all other controllers stopping, or just non ur controllers?
- Can two arms drivers be run on the same machine?

gavanderhoorn, 2020-09-24:
As a work-around, update this list:

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/936ff8d27eece1a87bcfd3fb26220487922a3704/ur_robot_driver/launch/ur_control.launch#L75-L80

dniewinski, 2020-09-24:
That list is the "do not murder" list?

dniewinski, 2020-09-24:
I've updated that list, but it doesn't seem to help.  I'm still seeing the same problem

gavanderhoorn, 2020-09-24:
Could you update your `ROSCONSOLE` config so it shows nodes and logger names?

It's not quite clear which node is printing those error messages.

dniewinski, 2020-09-24:
These are the errors from the first launch of the driver.  The second launch shows no errors
    [INFO] [/ros_control_controller_spawner]: Loading controller: joint_state_controller
    [INFO] [/ros_control_stopped_spawner]: Loading controller: pos_joint_traj_controller
    [INFO] [/ros_control_controller_spawner]: Loading controller: scaled_pos_joint_traj_controller
    [ INFO] [/ur_hardware_interface]: Calibration checked successfully.
    [ WARN] [/ur_hardware_interface]: No realtime capabilities found. Consider using a realtime system for better performance
    [ERROR] [/ros_control_stopped_spawner]: Failed to load pos_joint_traj_controller
    [INFO] [/ros_control_stopped_spawner]: Controller Spawner: Loaded controllers:
    [ERROR] [/ros_control_controller_spawner]: Failed to load scaled_pos_joint_traj_controller
    [INFO] [/ros_control_controller_spawner]: Loading controller: speed_scaling_state_controller
    [ INFO] [/ur_hardware_interface]: Negotiated RTDE protocol version to 2.
    [ INFO] [/ur_hardware_interface]: Setting up RTDE communication with frequency 125.000000
    [ERROR] [/ros_control_controller_spawner]: Failed to load speed_scaling_state_controller
    [INFO] [/ros_control_controller_spawner]: Loading controller: force_torque_sensor_controller
    [ INFO] [/ur_hardware_interface]: Robot ready to receive control commands.
    [ WARN] [/ur_hardware_interface]: No realtime capabilities found. Consider using a realtime system for better performance
    [ INFO] [/ur_hardware_interface]: Loaded ur_robot_driver hardware_interface
    [ INFO] [/ur_hardware_interface/ur_robot_state_helper]: Robot's safety mode is now NORMAL
    [ INFO] [/ur_hardware_interface/ur_robot_state_helper]: Robot mode is now RUNNING
    [ERROR] [/ros_control_controller_spawner]: Failed to load force_torque_sensor_controller
    [INFO] [/ros_control_controller_spawner]: Loading controller: robot_status_controller
    [INFO] [/ros_control_controller_spawner]: Controller Spawner: Loaded controllers: joint_state_controller, robot_status_controller

    [ERROR] [/husky_node]: This controller requires a hardware interface of type 'hardware_interface::PositionJointInterface'. Make sure this is registered in the hardware_interface::RobotHW class.
    [ERROR] [/husky_node]: Initializing controller 'pos_joint_traj_controller' failed
    [ERROR] [/husky_node]: This controller requires a hardware interface of type 'ur_controllers::ScaledPositionJointInterface'. Make sure this is registered in the hardware_interface::RobotHW class.
    [ERROR] [/husky_node]: Initializing controller 'scaled_pos_joint_traj_controller' failed
    [ERROR] [/husky_node]: This controller requires a hardware interface of type 'ur_controllers::SpeedScalingInterface'. Make sure this is registered in the hardware_interface::RobotHW class.
    [ERROR] [/husky_node]: Initializing controller 'speed_scaling_state_controller' failed
    [ERROR] [/husky_node]: This controller requires a hardware interface of type 'hardware_interface::ForceTorqueSensorInterface'. Make sure this is registered in the hardware_interface::RobotHW class.
    [ERROR] [/husky_node]: Initializing controller 'force_torque_sensor_controller' failed

fmauch, 2020-09-24:
Yes, as @gavanderhoorn wrote, the `controller_stopper` kills all controllers not whitelisted inside the `consistent_controllers` parameters. The intention is that the user cannot send commands to the robot while it isn't able to accept commands.

If you share the controller_manager namespace with other hardware devices, though, controllers from other drivers are killed, as well.

The same problem occurs when you want to load controllers. It seems that the controller_manager from the husky platform picks up the requests meant for the ur_robot_driver node and complains that it cannot load the controller, as it doesn't provide the appropriate hardware interfaces.

We usually handle this (and also do it like that on our husky platform) by starting the arm's driver inside a namespace. Something like this should be working:

```xml
  <group ns="ur_arm">
    <include file="$(find ur_robot_driver)/launch/ur_control.launch">
      <arg name="use_tool_communication" value="false"/>
      <arg name="controller_config_file" value="$(arg controller_config_file)"/>
      <arg name="robot_ip" value="$(arg robot_ip)"/>
      <arg name="kinematics_config" value="$(arg kinematics_config)"/>
      <arg name="tf_prefix" value="$(arg tf_prefix)"/>
      <arg name="controllers" value="$(arg controllers)"/>
      <arg name="stopped_controllers" value="$(arg stopped_controllers)"/>
      <arg name="headless_mode" value="$(arg headless_mode)"/>
    </include>

    <!-- As we are in a separate namespace, we need to run our own robot_state_publisher -->
    <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher"/>
  </group>
```

I don't know whether handling this differently would be preferable, this is the workflow I am familiar with.

gavanderhoorn, 2020-09-25:
Running the driver in a namespace was indeed also going to be my next suggestion.

But I wasn't sure whether he was using a combined hw setup or not.

dniewinski, 2020-09-25:
@gavanderhoorn @fmauch Thanks for the suggestion.  That seems like the simplest and cleanest solution.  I will have access to the hardware again on Monday and will definitely give it a try

dniewinski, 2020-09-28:
Namespacing the controller seems to be working.  Thanks for the help

closed 2020-09-28 by dniewinski