UniversalRobots/Universal_Robots_ROS_Driver #274: Movement cannot be completely cancelled after protective stop

agonzat, 2020-10-06:
# Summary

In a program when the robot enters the protective stop mode the current trajectory wants to be removed and the robot set to hold the current position.  For this the action interface (`scaled_joint_traj_controller/follow_joint_trajectory`) cancel topic is used. This does update the trajectory, but regardless of this, the robot will still perform a quick unwanted movement to a position nearby, along the trajectory it was following when it was stopped.

# Versions
 - ROS Driver version: latest
 - Robot Model: UR10e
 - Affected Robot Software Version(s): 5.9

# Impact
Cannot properly cancel a trajectory.

# Issue details
Things that I have checked (in `ScaledJointTrajectoryController`):
 - The cancel call gets through and does trigger the trajectory change (`cancelCB(...)` in controller base class `JointTrajectoryController`)
 - Current robot position at the time of creating the hold trajectory is correct
 - After sampling the trajectory, the `desired_joint_state_` is not equal to the current position. Instead it is a point along the trajectory it was following before the cancellation, very similar to the desired state it had before cancelling the trajectory. This makes the robot move abruptly into a position nearby the current position when the movement is resumed, then it stays there.
 - If the goal is not cancelled the robot will complete the whole trajectory when the stop is recovered (as intended, I believe).

## Use Case and Setup
In our use case when a protective stop is detected we want to make the robot restart holding the current position to then replan/perform other actions, not necessarily continue the current trajectory.

## Steps to Reproduce
1. Make the robot enter the protective stop state during a trajectory execution
2. Cancel the goal using the `scaled_joint_traj_controller/follow_joint_trajectory` action. (with `SimpleActionClient` for example);
3. Return the robot into normal operation (using dashboard interface)

## Expected Behavior
The robot doesn't move (holds position).

## Actual Behavior
The robot Moves to a nearby point along the cancelled trajectory on resume.

fmauch, 2020-10-07:
If I understood that correctly, this is actually "expected" behavior. (Expected as in this is how the controller currently works).

The scaled controller is basically only a trajectory controller where the update steps can be scaled down. What I think, what is happening:
 1. The robot goes to protective stop during a motion
 2. The trajectory gets cancelled. For the controller the robot is in the middle of the motion, it is currently simply not going forward, as speed_scaling is zero. In other words: The controller is sending the same setpoint in every control cycle. The actual "cancel" operation replaces the current trajectory with a ramp-down trajectory from the current setpoint (where the setpoint also contains velocities). The slope of this ramp is defined by the controller's `stop_trajectory_duration` parameter (see the [`joint_trajectory_controller` documentation](http://wiki.ros.org/joint_trajectory_controller).
 3. Once robot motion continues, the robot executes the stop ramp until it stands still.

If my assumption is correct, setting `stop_trajectory_duration` to 0 for example [here}(https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/config/ur10e_controllers.yaml#L44) for the ur10e should cancel this behavior.

However, I would agree that this is not the optimal behavior and should be considered a bug. If you could confirm my assumptions above, we can think about a proper solution for this. For example, we could use the robot's state instead of its current setpoint to create the stop trajectory.

agonzat, 2020-10-08:
The behavior seems to match what you explanation, the motion also takes roughly .5 seconds to complete. I will update the issue and add setting `stop_trajectory_duration` to 0 as a workaround for the issue.

I'm unsure about whether building the hold trajectory from the current robot state would be ideal, since it can make the robot abandon the trajectory in worse/riskier ways (than the current desired-state-based stop trajectory) if the motion is cancelled during jerky motions or when initially approaching the reference.

I will try and work a solution that initializes both trajectory builders (stop and hold) and decides during cancellation the strategy to use, depending on the current state.

closed 2021-08-03 by fmauch