UniversalRobots/Universal_Robots_ROS_Driver #275: Overshooting joint trajectory controller

felixvd, 2020-10-06:
# Summary

Our UR 5e overshoots significantly when executing this trajectory:

[image]

Sadly, I forgot to take a video of the trajectory in Rviz, but the joint states in the trajectory don't swing back like that. They move monotonously towards the final state (which the robot does arrive at, as well). I uploaded the trajectory in [this yaml file](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/5333937/ur5e_problem_trajectory_06_10_20.yaml.zip).

I would like to know if this behavior is expected and if there is a workaround. Thanks in advance for any insights.

# Versions
 - ROS Driver version: commit 3ce362
 - Affected Robot Software Version(s): 5.7
 - Affected Robot Hardware Version(s): UR 5e
 - Robot Serial Number: -
 - UR+ product(s) installed: - 
 - URCaps Software version(s): 1.0.2

# Impact
The robot does not stop where it is supposed to.

## Use Case and Setup
We need the robot to follow the trajectory without overshooting this much, as it would break our tools. But we do need the robot to move at least this quickly.

## Steps to Reproduce

You can move the robot to the start state of this trajectory*, and then execute it by sending it to `/ur_hardware_interface/follow_joint_trajectory/command` : 
[ur5e_problem_trajectory_06_10_20.yaml.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/5333937/ur5e_problem_trajectory_06_10_20.yaml.zip)

*The start of the trajectory is `[2.981921261631558, -1.27472656822998, 1.3204057526082944, -1.6164710215873699, 3.075542515700016, -6.91763185490812e-07]`.

## Expected Behavior
The robot should follow the trajectory without overshooting.

## Actual Behavior
The overshoot is significant.

## Workaround Suggestion
I can try executing the same trajectory slower, but that won't help us (the speed is necessary for our application). Converting the trajectory to a URscript and sending it to the robot controller box could work. If you have any code to share that would be great.

felixvd, 2020-10-07:
I found [this `do_extrapolate` line](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/ros_control.urscript#L51) in the urscript running on the robot that seems to continue applying joint commands until it receives a new one (is this safe?). To make sure that's not the culprit, I set the value in that line of `ros_control.urscript` to `False`, activated headless mode, and removed the URCap from my robot to be sure that it is the script that is being executed on the robot. Sadly, the problem persists.

I also confirmed that slowing down the trajectory fixes the overshooting (unsurprisingly), but since we need to move and stop quickly, I hope to find a better solution.

fmauch, 2020-10-07:
My first question would be whether the robot is actually able to execute the fast motion or whether speed scaling is a factor in that execution.You could have a look at the topics `speed_scaling_factor` and `pos_joint_traj_controller/state` to find out more. I've seen such overshoots by the `joint_trajectory_controller` before, when non-realistic constraints are set for the setpoints (e.g. accelerating by a large portion for a very short way. This should show in the `pos_joint_traj_controller/state/desired` field if this is the case. A plot of a relevant axis would be interesting to investigate this further.

felixvd, 2020-10-09:
We don't touch speed scaling and the pendants are set to full speed.

This is the plot of [this trajectory](https://gist.github.com/felixvd/1131e2fee68552622df29376bd71468f#file-ur5e_problem_trajectory_2-yaml) (technically not the same as in the video, but essentially):

[image] 

It's true and visible in the video that the elbow joint puts in a lot of effort, but it doesn't seem extreme in the plot. I guess there is no official value for the maximum acceleration values. Maybe we will just have to slow down the relevant parts experimentally.

gavanderhoorn, 2020-10-09:
> We don't touch speed scaling and the pendants are set to full speed.

The question is not whether *you* change the scaling, but whether the safety system in the robot controller does.

felixvd, 2020-10-09:
I just used `yaml.dump` on the msg object. Is there a better way? The `rospy` documentation is hard to get toright now because the `ros.org` domain seems to have a problem.

I don't know if the robot could handle the trajectory in URScript - I haven't tesated it. Is there an example for how to convert the trajectory to a URScript? I couldn't find one or think of a simple solution.

fmauch, 2020-10-12:
> I just used yaml.dump on the msg object. Is there a better way? The rospy documentation is hard to get to right now because the ros.org domain seems to have a problem.

I usually use a `rostopic echo -n 1 <topic_name> > output.yaml` to record messages. Also, you recorded the `command` topic, but the `state` topic would be the interesting one, as we can see the desired and actual state there.

> I don't know if the robot could handle the trajectory in URScript - I haven't tested it. Is there an example for how to convert the trajectory to a URScript? I couldn't find one or think of a simple solution.

I'm not sure what a proper conversion would look like as it is hard to translate the spline-interpolated trajectory into waypoint+blending radius notation that is being used for the URScript trajectory.

> There is a little dip. I guess that means the trajectory wasn't considered safe?

At least that means that the "scaling" part of the controller is being used. The robot controller slowed down the commanded motion due to the safety constraints configured on the robot controller.

felixvd, 2020-10-15:
I see. I didn't know the state topic included that information.

We ended up modifying the motion to avoid the elbow acceleration, although we would have preferred a generally applicable solution to generate safely executable trajectories. Sadly, neither the time parametrization methods we had available out of the box in MoveIt nor slowing down only the sections with high accelerations worked well for us at all.

> I'm not sure what a proper conversion would look like as it is hard to translate the spline-interpolated trajectory into waypoint+blending radius notation that is being used for the URScript trajectory.

Me neither, but I'd be very interested in even an imperfect conversion :)

closed 2020-10-15 by felixvd

DavidYaonanZhu, 2022-07-14:
So the solution is to slow down the trajectory?