UniversalRobots/Universal_Robots_ROS_Driver #276: Action client cancel goal does not immediately stop movement

sharpe-developer, 2020-10-07:
# Summary

When an active goal is cancelled the robot does not stop immediately. However it looks like the arm continues through some or all of the remaining trajectory in the goal. Is this the expected behavior?

Note: my start and end pose only differ by approximately 5cm so the total distance for the goal is small. 

# Versions

- Ubuntu 18.04.3 with Real Time patch installed
- ROS Melodic
- ROS Driver version: 0.0.3
- Affected Robot Software Version(s): 5.9.1.1031110
- URCaps Software version(s): external control 1.0.4, rs485 1.0.0

## Steps to Reproduce

I am using `/scaled_pos_joint_traj_controller/follow_joint_trajectory`

I command a move (without wait) using simple action client. The goal is moving the robot pose on the X axis for 5 cm.
`self.action_client.send_goal(goal)`

While the robot is moving I am monitoring a discrete input. This input goes high approximately 2cm into the move. When the discrete goes high I cancel the goal.
`self.action_client.cancel_goal()`

I assumed the arm would halt very close to the position when the discrete is asserted. However it looks like the arm continues  through some or all of the remaining trajectory in the goal.

## Expected Behavior

I expect the robot to freeze at (or very close to) the pose the cancel goal was issued.

## Actual Behavior

The robot continues moving for a couple seconds after the cancel goal was issued and the TCP position goes well beyond the TCP position when cancel goal was issued

gavanderhoorn, 2020-10-08:
Please take a look at #274. I believe this may have a similar underlying cause.

sharpe-developer, 2020-10-09:
Yes, you are correct. I looked at #274 but missed the connection between protective stop and a canceled goal.

Setting `stop_trajectory_duration` to 0 in `ur*_controllers.yaml` fixes the issue.

closed 2020-10-09 by sharpe-developer