UniversalRobots/Universal_Robots_ROS_Driver #277: The robot moves slightly after switching the controller from velocity to trajectory controllers

akihikoy, 2020-10-12:
# Summary
The robot moves slightly after switching the controller from velocity to trajectory controllers (`joint_group_vel_controller` to `scaled_pos_joint_traj_controller`).

# Versions
 - ROS Driver version: 936ff8d27eece1a87bcfd3fb26220487922a3704
 - URCaps: externalcontrol-1.0.4.urcap
 - Robot: UR3e

# Issue details
I'm testing a joint velocity controller `joint_group_vel_controller` with simple commands.
I actually followed https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/231 by using the script https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/231#issuecomment-688415091 , and it worked.
I changed the target velocities to move the whole robot joints, and it also worked.
Then I embedded the control switching commands into the program where the above issue happened.

The entire program is here:

```
#!/usr/bin/env python

import roslib
import rospy
import std_msgs.msg
import controller_manager_msgs.srv
import math,time,sys

if __name__=='__main__':
  rospy.init_node('velocity_control_test', disable_signals=True)  #NOTE: for executing the stop motion commands after Ctrl+C.
  pub= rospy.Publisher('/joint_group_vel_controller/command', std_msgs.msg.Float64MultiArray, queue_size=10)
  msg= std_msgs.msg.Float64MultiArray()
  msg.data= [0.0]*6
  msg.layout.data_offset= 1

  srv_sw_ctrl= rospy.ServiceProxy('/controller_manager/switch_controller', controller_manager_msgs.srv.SwitchController)
  sw_ctrl_req= controller_manager_msgs.srv.SwitchControllerRequest()
  sw_ctrl_req.strictness= sw_ctrl_req.STRICT
  sw_ctrl_req.stop_controllers= ['scaled_pos_joint_traj_controller']
  sw_ctrl_req.start_controllers= []
  srv_sw_ctrl(sw_ctrl_req)
  sw_ctrl_req.stop_controllers= []
  sw_ctrl_req.start_controllers= ['joint_group_vel_controller']
  srv_sw_ctrl(sw_ctrl_req)

  #rate= rospy.Rate(125)  #UR-CB receives velocities at 125 Hz.
  rate= rospy.Rate(500)  #UR-e receives velocities at 500 Hz.
  t0= time.time()
  try:
    while not rospy.is_shutdown():
      t= time.time()-t0
      msg.data= [0.05*math.sin(math.pi*t)]*6
      pub.publish(msg)
      rate.sleep()
  except KeyboardInterrupt:
    print 'Interrupted'
  finally:
    print 'Stopping the robot...'
    msg.data= [0.0]*6
    pub.publish(msg)

    #WARNING: Without this loop or the sleep command, the robot moves slightly after switching the controller.
    for i in range(10):
      pub.publish(msg)
      rate.sleep()
    #rospy.sleep(0.05)

    sw_ctrl_req= controller_manager_msgs.srv.SwitchControllerRequest()
    sw_ctrl_req.strictness= sw_ctrl_req.STRICT
    sw_ctrl_req.stop_controllers= ['joint_group_vel_controller']
    sw_ctrl_req.start_controllers= []
    srv_sw_ctrl(sw_ctrl_req)
    sw_ctrl_req.stop_controllers= []
    sw_ctrl_req.start_controllers= ['scaled_pos_joint_traj_controller']
    srv_sw_ctrl(sw_ctrl_req)
```

In the last block, I'm switching the controller from `joint_group_vel_controller` to `scaled_pos_joint_traj_controller`.
In above program, a loop starting with `for i in range(10):` is introduced to avoid the issue I am reporting.
If this block is commented out, the robot moves slightly after switching the controller in the last block.

By using `rospy.sleep(0.05)` instead of the `for i in range(10):` block, the robot also stops.

Even if both `for i in range(10):` block and `rospy.sleep(0.05)` are commented out, the robot stops if I **do not switch the controller** from `joint_group_vel_controller` to `scaled_pos_joint_traj_controller` (i.e. commenting out the last block).

So, I thought there might be an issue in controllers.  Or, please point out if there is something wrong with my attempt.

Many thanks!
Akihiko

fmauch, 2020-11-15:
I think, I do not fully understand the behavior that you observe. What exactly do you mean by "the robot moves slightly after switching the controller in the last block"?

stefanscherzinger, 2021-08-09:
@akihikoy Is this still relevant and do you still observe this behavior?

stefanscherzinger, 2021-09-01:
Feel free to re-open if this issue persists.

closed 2021-09-01 by stefanscherzinger