UniversalRobots/Universal_Robots_ROS_Driver #279: Faster Position Control?

w-kyle, 2020-10-13:
# Summary
Is there any way to achieve a fast position controller response with this driver? I am operating in a tele-oeration setting, where position trajectories are not useful, because the desired position is not known more than one loop iteration in advance. Right now I am experiencing slow position control response while publishing joint positions on the `/scaled_pos_traj_controller/command` topic (the same technique that is used by the rqt_joint_trajectory_controller). The commanded joint responds quickly at first (if the target position is relatively far), but then slows gradually and takes a relatively long time to reach the target joint position. The target position can be reached much faster using a single servoj command. It seems like the position trajectory controller is interpolating a smoothed spline trajectory that results in slow movement near the desired position. Is there a way to more directly control joint positions, without streaming script commands, as that option seems to be discouraged in other issues?

# Versions
 - ROS Driver version: Most Recent Master, also affects older versions
 - Affected Robot Software Version(s): 
 - Affected Robot Hardware Version(s): UR5e
 - Robot Serial Number: 20195501039
 - UR+ product(s) installed: None
 - URCaps Software version(s): 1.0.4

# Impact
I experience slow, laggy movements when controlling the arm in a tele-operation setting, where a user manipulates a control arm, and the UR5e mirrors the movement. This happens when using either of the position based controllers, with speed scaling set to 100%. See the plot below for an example.

# Issue details
I am controlling a UR5e in a tel-operation setting, with a master control arm that matches the UR5e kinematics. This setup makes it possible to work directly in the joint space, in a position control mode. In this setting, I would like to directly command the joint positions, because our control arm encoders output position, and I want to avoid implementing a velocity based controller to control position on top of the ROS driver. I do not know trajectories in advance, so planning full trajectories is useless. Is it possible to use one of the position control modes to do this? My current strategy is to stream positions on the `/scaled_pos_traj_controller/command` topic, which allows me to avoid specifying and canceling lots of trajectories, but it seems that a spline interpolated trajectory is still set as the goal state. Below is a plot showing an exaggerated step function input to the controller for a single joint.
[image]
In this example, the position initially lags the and maxes out the velocity limit, then at the end of the trajectory the joint slows dramatically and takes an extra 2 seconds to reach the target position. With smaller positions steps, the lagging becomes more pronounced, and makes it difficult for a user to precisely move the arm. For comparison, if you send a single servoj command you can get a trajectory that looks more like this (note that I was playing with the safety settings, so the max speed is different):
[image]
Ideally I would like the position output to look more like the output from the servoj command. Is it possible to do this? So far I have had trouble finding documentation on using the position controllers in this way.

## Project status at point of discovered
During initial proof of concept testing.

## Steps to Reproduce

1. Start the ROS driver and the External Control program.
2. Run the rqt_joint_trajectory_controler gui
3. Command a step input by typing in a new position for one joint. ~180 degrees is plotted above
4. Plot the` /scaled_position_traj_controller/command` and `joint_states` position and velocity topics with Plotjuggler, or the RQT utility and observe the lag. 

## Expected Behavior
Ideally I would expect low level position control, with no mediated trajectroy.

## Actual Behavior
There seems to be an imposed smooth, slow trajectory, over my position inputs. 

## Workaround Suggestion
At present I can get around this by streaming servoj commands, but that is not a good solution.

gavanderhoorn, 2020-10-13:
For tele-operation you'll want to use any of the `JointGroup*` controllers, not the `JointTrajectoryController`.

You're not executing trajectories.

fmauch, 2020-10-13:
Just as a note: You will get the most responsive behavior using the [`joint_group_vel_controller`](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/config/ur5e_controllers.yaml#L129). You can define a position controller analog to this if you definitely want to do position control. However, be aware that the URScript [servoj command](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/ros_control.urscript#L69) will also do a little bit of interpolation and therefore add a little bit of delay to your command.This is why I would recommend using a velocity-based controller in case of direct joint control.

w-kyle, 2020-10-13:
Ok, thanks, this is helpful. Yes, I agree this is not an issue with the driver.

closed 2020-10-13 by w-kyle

gavanderhoorn, 2020-10-13:
There could still be issues underlying this though, but at least the higher-level issue you raised was not directly something I would categorise as an issue with the driver.