UniversalRobots/Universal_Robots_ROS_Driver #280: Unable to run URCap on UR3 Teach Pendant - "Reason: connect timed out"

ajaydxb97, 2020-10-28:
# Summary

This is for the UR3 robot. I successfully built the driver in my workspace and installed the URCap on the UR3's teach pendant as given in the tutorial. But when I start the external control program on my teach pendant after connecting it to my laptop via the LAN cable and launching the driver, I get an error on the pendant saying: "The connection to the remote PC at 192.168.1.105:50002 could not be established. Reason: connect timed out". (Display of error attached)

[image]

I made sure the IP addresses of the robot as well as my laptop were correct. I checked the Polyscope version and made sure it was at least 3.7. 
I know for sure that there are no connection issues because I was able to ping the robot from my laptop as well as control it using the URX python library.

# Versions
 - OS: Ubuntu 18.04 LTS inside VMWare Workstation Player
 - ROS version: Melodic
 - UR Robot: UR3
 - UR Driver version: 936ff8d27eece1a87bcfd3fb26220487922a3704
 - URCaps Software version: externalcontrol-1.0.4.urcap

Any help would be greatly appreciated. Thanks!

fmauch, 2020-10-28:
* Did you run the driver on the ROS pc before starting the program on the teach pendant?
 * Could you please elaborate your IP address setup? What address is configured on the teach pendant / the ROS pc?

ajaydxb97, 2020-10-29:
> * Did you run the driver on the ROS pc before starting the program on the teach pendant?
> * Could you please elaborate your IP address setup? What address is configured on the teach pendant / the ROS pc?

1) Yes, I started the driver on the ROS pc and before starting the external control program on the teach pendant.
2) ROS PC IP: 192.168.1.105 (static IP address),    UR3/Teach Pendant IP: 192.168.1.50 (already set in teach pendant and can be viewed in its 'About' dialog box)

gavanderhoorn, 2020-10-29:
Can you check whether you have a firewall enabled on the network connection which connects to the control box?

Also: please copy-paste the `roslaunch` command you use to start the driver and the output which follows verbatim into a comment here. Please make sure to do that [in a code block](https://docs.github.com/en/free-pro-team@latest/github/writing-on-github/creating-and-highlighting-code-blocks).

ajaydxb97, 2020-11-21:
Ok, so I finally figured out why I couldn't connect the UR3 driver on my laptop with the real UR3. It was a Virtual Machine issue and a few more things. Now I am able to move the robot using the driver via Moveit.

Solution: 
1) In Virtual Machine Settings, change Network Adapter to: Bridged (Automatic) (It was previously on NAT). Check the box that says 'Replicate physical network connection state'. In the Configure Adapters dialog box, uncheck everything except the Ethernet option. 
2) Make sure the IP address of the host is static and not automatic DHCP. (This was already the case)

closed 2020-11-21 by ajaydxb97