UniversalRobots/Universal_Robots_ROS_Driver #281: usage Universal_Robots_ROS_Driver with rviz or gazebo

hariharan382, 2020-11-01:
i know this may be dumb question but i want to know whether ur robots  can be simulated by this driver "Universal_Robots_ROS_Driver" because ur_driver in previous package was depracated can you please anyone help me to simulate the ur 10  with the joint state coordinates

fmauch, 2020-11-06:
You can use the URSim simulation available on UR's website in order to simulate a robot. With that simulation you can use this driver as if a real robot was connected.

Of course, you can also use the gazebo simulation from [ur_gazebo](https://github.com/ros-industrial/universal_robot/tree/melodic-devel-staging/ur_gazebo) that doesn't need any interaction of this driver.

hariharan382, 2020-11-06:
sir i want to control to with python node but there is already python script called "test_move.py" but facing waiting for server i saw all similar issues but none solved my problem

fmauch, 2020-11-16:
> sir i want to control to with python node but there is already python script called "test_move.py" but facing waiting for server i saw all similar issues but none solved my problem

I don't see how that should be related to the original question. Also note that this driver does not contain a `test_move.py` script.

fmauch, 2021-05-15:
Closing this due to no further updates. Feel free to reopen if updates arise.

closed 2021-05-15 by fmauch