UniversalRobots/Universal_Robots_ROS_Driver #282: Can't install dependencies on Noetic when building from source

tkelestemur, 2020-11-04:
# Summary
The command for installing dependencies throws an error when building from source.
Below is the command and the error I get:

```
➜  ros_ws rosdep install --rosdistro noetic --from-paths src --ignore-src -y

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur_robot_driver: Cannot locate rosdep definition for [industrial_robot_status_controller]
```

# Versions
 - ROS Driver version: `master`
 - Affected Robot Software Version(s): UR Software 5.3.1
 - Affected Robot Hardware Version(s): UR5e
 - Robot Serial Number: N/A
 - UR+ product(s) installed: N/A
 - URCaps Software version(s): External Control 1.04
- OS: Ubuntu 20.04
- ROS: Noetic

gavanderhoorn, 2020-11-05:
The readme does not mention Noetic and there is a soft requirement to use Melodic (in the [Requirements section](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/e239b638ade4a2894d5d5772502826b772846adb/README.md#requirements)).

However, that dependency should not be too difficult to release to Noetic (I'm the maintainer of that package).

It would be good to know whether that's the only dependency which fails, so please re-run your `rosdep` command but this time use `check` instead of `install` for the verb.

tkelestemur, 2020-11-05:
I managed to install the UR ROS driver by building [industrial_core](https://github.com/ros-industrial/industrial_core) and your package [industrial_robot_status_controller](https://github.com/gavanderhoorn/industrial_robot_status_controller) from source. 

It seems like the these two packages are not available as binary thus rosdep was not able to install. The source of the problem is that industrial_core is blocked by the KDL library which recently changed [rosdep keys](http://wiki.ros.org/noetic/Migration#Use_Orocos_KDL_and_BFL_rosdep_keys). The industrial_core package needs to reflect those changes to be available as binary. 

Here is the output of the `rosdep check`

```
rosdep check ur_robot_driver
System dependencies have not been satisfied:
apt	socat
ERROR[ur_robot_driver]: Cannot locate rosdep definition for [ur_msgs]
	rosdep key : ur_msgs
	OS name    : ubuntu
	OS version : focal
	Data:
<no data>
```

gavanderhoorn, 2020-11-05:
Please run `rosdep check` with all the arguments you gave to `rosdep install`. Do not run it against the `ur_robot_driver` package.

I was already aware of the cause of why you couldn't install `industrial_robot_status_controller`.

tkelestemur, 2020-11-05:
Oh okay. Here you go:
```
rosdep check --from-paths src --ignore-src -y
System dependencies have not been satisfied:
apt	ros-noetic-effort-controllers
apt	socat
ERROR[ur_robot_driver]: Cannot locate rosdep definition for [industrial_robot_status_controller]
	rosdep key : industrial_robot_status_controller
	OS name    : ubuntu
	OS version : focal
	Data:
<no data>
```

gavanderhoorn, 2020-11-05:
That's whatI would expect.

As soon as the release of `industrial_msgs` gets merged, I'll release `industrial_robot_status_controller` and everything should work again.

gavanderhoorn, 2020-11-05:
Everything's been released.

If you configure your system to use the testing repository you should be able to install the dependencies using just the released packages again.

tkelestemur, 2020-11-05:
Just changed to `ros-testing`, ran `sudo apt update` and `rosdep update`. It's able to locate it but can't install:
```
rosdep install --from-paths src --ignore-src -y
executing command [sudo -H apt-get install -y ros-noetic-industrial-robot-status-interface]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-noetic-industrial-robot-status-interface
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-noetic-industrial-robot-status-interface] failed
```

gavanderhoorn, 2020-11-05:
Well I would give the buildfarm some time to actually *build* the package.

tkelestemur, 2020-11-06:
I tried again and it worked without any problems. Thanks for helping @gavanderhoorn!

closed 2020-11-06 by tkelestemur