UniversalRobots/Universal_Robots_ROS_Driver #283: Problem connecting Driver over Wifi

krauluk1, 2020-11-05:
# Issue
Hi,

I am trying to connect my UR3e with my computer over WIFI. 

Therefor, I installed this package, made a MoveIt config etc. and if I connect my computer to the fritzbox, it works pretty well. But when trying to connect over wifi, it works fine till trying to open RVIZ. The robot kind of loses connection (Changes from Running to Stopped on Teachpannel). Rviz remains open and there is no bug report about what went wrong.

# Versions
 - ROS Driver version: latest
 - Affected Robot Software Version(s):  Universal Robot 5.8
 - Affected Robot Hardware Version(s): 
 - Robot Serial Number:
 - UR+ product(s) installed: External Control, rs485
 - URCaps Software version(s):

# Impact
Cannot plan over WIFI with RVIZ / Can not Move the Robot over WIFI

# Issue details
When starting RVIZ, Robot loses connection (or something like that). No error message recieved. Setup works well when connecting laptop with LAN.

## Use Case and Setup
I want to move the robot with rviz over wifi like I do when connected over LAN.
(Planning with moveit rviz)

## Project status at point of discovered
Communication over LAN works.

Trying to do the same over wifi (just changed the ip on robot and script in "Steps to Reproduce").

## Steps to Reproduce
My steps when starting the controller
```
<launch>
    <arg name="limited"  value="false"/>
    <arg name="robot_ip"  value="192.xxx.xxx.xx"/>
    
    <!-- Launch Bringup for Connection-->
    <include file="$(find ur_robot_driver)/launch/ur3e_bringup.launch">
        <arg name="robot_ip" value="$(arg robot_ip)"/>
        <arg name="limited" value="$(arg limited)"/>
    </include>
    <!-- -->
    
    <arg name="node_start_delay" default="4.0" />  

    <!-- Move Group launch -->
    <arg name="sim" default="false" />
   
    <!-- Remap follow_joint_trajectory -->
    <remap if="$(arg sim)" from="/follow_joint_trajectory" to="/arm_controller/follow_joint_trajectory"/>
  
    <!-- Launch moveit -->
    <include file="$(find robot_moveit_config)/launch/move_group.launch">
        <arg name="publish_monitored_planning_scene" value="true" />
        <arg name="lpre" value="bash -c 'sleep $(arg node_start_delay); $0 $@'" />
    </include>
    <!-- -->

    <!-- Moveit Rviz.launch-->
    <include file="$(find robot_moveit_config)/launch/moveit_rviz.launch" >
        <arg name="lpre" value="bash -c 'sleep $(arg node_start_delay); $0 $@'" />
    </include>
    <!-- -->

    <!-- GRIPPER -->
    <!-- Start node for gripper -->
    <node pkg="robotiq_2f_gripper_control" name="Robotiq2FGripperRtuNode" type="Robotiq2FGripperRtuNode.py" respawn="true" output="screen"/> 
</launch>
```
1. Launch this controller
2. Start move group
3. Launch Moveit
3. Launch Moveit Rviz <-- Here is the failture (when not included / no Rviz opened, no connection lost)

## Expected Behavior
Rviz opens and I can go on planning. (like I do when connecting over LAN)

## Actual Behavior
RVIZ opens. Robot changes from Running to Stopped. 

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

gavanderhoorn, 2020-11-05:
This is just a fellow user commenting, but I would not expect this to work over wifi. The UR controller sends (and sort-of expects) data and setpoints at 500 Hz (in your case, with an e-Series controller). As any control application, stability of the interface is very important and has a large impact on performance (motion, jitter, smoothness, etc).

Basic/consumer-grade wifi cannot guarantee a good enough connection, so what you describe is something I would expect.

Work is underway to implement additional control modes which will at least for the execution of motion be a bit more robust against lossy network connections, but those modes will support different use-cases than the current implementation (ie: reactive control vs execution of complete trajectories on the UR controller).

But even then I would suggest to really look at using a wired connection, as it'll always give you the best UX when controlling a robot externally.

fmauch, 2020-11-06:
Nothing to add from my side to @gavanderhoorn's comment. I'll close this, as it has been sufficiently answered. Feel free to drop additional comments or reopen if you disagree.

closed 2020-11-06 by fmauch