UniversalRobots/Universal_Robots_ROS_Driver #284: cmake error

hariharan382, 2020-11-06:
CMake Error at Universal_Robots_ROS_Driver/ur_calibration/CMakeLists.txt:12 (find_package):
  By not providing "Findur_client_library.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ur_client_library", but CMake did not find one.

  Could not find a package configuration file provided by "ur_client_library"
  with any of the following names:

    ur_client_libraryConfig.cmake
    ur_client_library-config.cmake

  Add the installation prefix of "ur_client_library" to CMAKE_PREFIX_PATH or
  set "ur_client_library_DIR" to a directory containing one of the above
  files.  If "ur_client_library" provides a separate development package or
  SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/hari/ur10_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/hari/ur10_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

hariharan382, 2020-11-06:
i have no idea to solve this error please help

fmauch, 2020-11-06:
Probably you have missed installing the `rosdep install` step from the [build instructions](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building)

eibow1012, 2020-11-22:
I got a similar error when using the GCC6 compiler.
I was able to make it successfully using the GCC7 compiler.

fmauch, 2020-11-22:
The error mentioned above is not a gcc but a cmake error.

eibow1012, 2020-11-23:
I'm sorry.
It was. But was it a coincidence that switching from the GCC6 environment to the GCC7 environment worked?
For the time being, the result went well.
It wasn't helpful.

SCh-J, 2021-03-10:
@hariharan382 Did you find a solution for the issue? I'm facing the same issue and am really stuck right now.

fmauch, 2021-03-10:
@SCh-J could you please post exactly what you have been doing and what the output of that is?

SCh-J, 2021-03-10:
It seems to work with the all-source build.

fmauch, 2021-03-11:
@SCh-J If you could still share the output of your problematic workspace this might help debugging what was going on.

Koriath, 2021-03-24:
Hello, I'm having a similar problem.
I've attempted the normal build instructions, and got the same error as @hariharan382.
Following @SCh-J , I tried to do the all source build. For the most part, it works right until it has to build "ur_client_library".
It throws this error:

`-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Boost: C:/opt/ros/melodic/x64/include/boost-1_66 (found version "1.66.0")
-- Performing C++11 Test
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Failed
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Failed
CMake Error at CMakeModules/DefineCXX11CompilerFlag.cmake:50 (message):
  The compiler C:/Program Files (x86)/Microsoft Visual
  Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x64/cl.exe has
  no C++11 support.  Please use a different C++ compiler.
Call Stack (most recent call first):
  CMakeLists.txt:18 (DEFINE_CXX_11_COMPILER_FLAG)`
  
  It seems to be checking for c++11 support. I have visual studio installed and developed other applications with it, but then remembered that, maybe I only used c++17 features? Maybe I didn't have c++11 installed or something close to it.
  So I begun reading "ur_client_library" repo and found that only the "boost" branch uses c++11.
  I tried to clone the main branch, but I got a similar error, this time complaining about lack of c++17 support.
  
  One thing of note is that rosdep -install in the readme doesn't appear to be working:
  `C:\catkin_ws>rosdep install --from-paths src --ignore-src -y
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur_robot_driver: No definition of [socat] for OS [windows]`

I'm not sure if this intereferes with anything, but from my (granted, incomplete) point of view, socat isn't the issue here.
However, if rosdep was suposed to install anything else... it doesn't. And I'm not entirely sure why.

Related to this, one thing I found on [StackOverflow](https://stackoverflow.com/questions/49093833/cmake-3-10-not-working-at-all-with-visual-studio-2015-2017) seems to mention that the scripts installing/making/building the library should not be checking C++ version but rather letting CMAKE handle it.

Is there anything I can do on my end? Because the older drivers are deprecated and no longer on my UR (v3.15) and these new one''s don't compile/build at all.

fmauch, 2021-03-24:
@Koriath It looks as you are trying to build this on Windows, which is not working. Currently, the `ur_client_library` will only work on Linux.

Koriath, 2021-03-24:
@fmauch So I'm proper screwed then. I'm working on a company PC, my requests to install linux were denied several times now.
the older drivers compiled on windows but with the new firmware, the deserialization script broke communication with the robot and no longer works. Would you know of anyway to get this thing working? I'm mostly interested in simply extracting the position of the robot and exposing it to another app someone is developing (I believe something made in unity, a robot visualizer of sorts).

fmauch, 2021-03-24:
@Koriath You could write a Windows socket implementation for the library, handle all threading stuff separately and update the CMake scripts to determine the required c++ standard to make them MSVC compatible. Those would be the missing things that I can think of out of my head. If you'd like to implement that, we would be happy to review a PR on this :-)

Koriath, 2021-03-24:
@fmauch It would be pretty hard to justify it to my boss. He doesn't pay me to write open-source software XD
Maybe in my spare time, But that will be finished to late for the project I'm in. Thanks for the swift answers anyway.

gavanderhoorn, 2021-03-24:
If you're not necessarily required to use `ur_robot_driver`, you could see whether [ur_rtde](https://sdurobotics.gitlab.io/ur_rtde) works for you. It [seems to support building on Windows](https://sdurobotics.gitlab.io/ur_rtde/installation/installation.html#windows).

urrsk, 2021-03-24:
Maybe this RTDE implementation could be useful:  https://www.universal-robots.com/articles/ur/interface-communication/real-time-data-exchange-rtde-guide/

closed 2021-08-03 by fmauch