UniversalRobots/Universal_Robots_ROS_Driver #289: Which is best approach to control UR5 CB3 in ROS?

ManueU, 2020-11-16:
Hi, I am using ROS noetic and I'd like to teleoperate an UR5 CB3. I have some questions:

1. I read @gavanderhoorn 's comment to this issue #122 , in which he suggests two different approaches:
 
> So for individual points:
> use the command topic of the JointTrajectoryController, or
> switch to a JointGroupVelocityController or JointGroupPositionController and publish to their command topic
> For trajectories:
> create an action client and submit control_msgs/FollowJointTrajectory action goals
> note that the JointTrajectoryController also supports trajectory replacement. 

I wonder what is the difference between JointTrajectoryController, JointGroup* and FollowJointTrajectory:
   - Could it be related to the smoothing of the trajectory?
   - Could it be related to the frequency? In particular: Could the reading frequency from topic be different between controllers? And, It is different from 125 Hz (communication frequency with the UR5)? If one of the previous assumption is true, I'd like to understand the specific reason. 

2. I managed to install the UR ROS driver and make it works with rqt_controller addressing /scaled_pos_joint_traj_controller/command. Looking at the rqt_graph, I understand that the topic connected to JointTrajectoryController is /scaled_pos_traj_controller/command and the topic connected to JointGroupVelocityController is /joint_group_vel_controller/command. But I can't see a topic related to the JointGroupPositionController. Is it because I am supposed to define it in case I want to use that? 
I read a comment by @fmauch  in the issue #172 
> Inside the driver we use ROS-Control to implement different controllers. See its documentation for details.

 I read also the documentation about ros_control and ros_controllers? Can I implement the JointGroupPositionController using one of that?

3. Instead of using the controller's topics to send the goal positions to the robot, I initially thought about merely sending the cartesian pose as move* input via URScript, but then I realized in the issue #162  @fmauch  warns not to use this approach and, in the same issue,  @gavanderhoorn  says that ROS/ur_robot_driver is not the appropriate interface to do this. What is the specific reason you wouldn't suggest that? You said the driver is not the best interface to do that, which would be the right interface to do that? I would prefer using ROS because the other devices of the setup are already integrated in ROS. Which would be the drawbacks of such a method? Jerky movements, communication latencies, and, in case, why?

## Use Case and Setup
Teleoperation setup

fmauch, 2020-11-16:
1. See the respective controller documentations for the [`ForwardJointGroupCommandController`](https://github.com/ros-controls/ros_controllers/blob/noetic-devel/forward_command_controller/include/forward_command_controller/forward_joint_group_command_controller.h#L54-L70) and the [joint_trajectory_controller](http://wiki.ros.org/joint_trajectory_controller) for details.
2. As written in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/172#issuecomment-626492559 you'll have to define it inside the [`ur5_controllers.yaml`](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/config/ur5_controllers.yaml) file similar to the `joint_group_vel_controller`. Once done, you'll be able to activate it using the [`controller_manager`](http://wiki.ros.org/controller_manager).
3. As written in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/162#issuecomment-617774745 you won't receive any feedback from the robot using this approach. Also, each time script code gets sent to the robot it gets compiled into executable code which takes a non-deterministic amount of time. Hereby, you'll introduce latency to the communication.
My colleagues are currently working on implementing Cartesian control for ros_control which will also contain a controller similar to the `ForwardJointGroupCommandController` that can be used for Cartesian servoing. I've used a prototype of this to steer the robot using a spacemouse. Maybe @stefanscherzinger can say more about the state of this.

ManueU, 2020-11-18:
Hi,  thank you for your fast reply. 

1. I have understood that in order to use the joint_trajectory_controller, I have two possibilities to send trajectories: using the action interface (/scaled_pos_joint_traj_controller/follow_joint_trajectory) or using the topic interface (/scaled_pos_joint_traj_controller). Is it correct?
But, I can't understand yet the difference between JointGroupPositionController and JointTrajectoryController. Can you explain it to me?

I'd like to clarify some frequency questions.
I read that the control loop in ur_hardware_interface runs as fast as possible. 
As a consequence, I wonder how the frequency of my architecture is managed, from the controller to the robot. In particular, I am asking if I have to use a specific frequency to publish goal positions on the followings (and the reason for that): 
/scale_pos_joint_traj_controller/command
/joint_group_pos_controller/command
/scaled_pos_joint_traj_controller/follow_joint_trajectory

stefanscherzinger, 2020-11-18:
> My colleagues are currently working on implementing Cartesian control for ros_control which will also contain a controller similar to the ForwardJointGroupCommandController that can be used for Cartesian servoing. I've used a prototype of this to steer the robot using a spacemouse. 

The timeline of this is a little split-up. There will be some functionality publicly available [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control) soon (as in few days) and a complete release later (as in few months). There will be continuous updates in between. The first one should get you going with using ROS-control as you normally would with a Cartesian-based pipeline, however, not (yet) providing the rich set of ROS-controllers.

@ManueU If you need teleoperation now, let me point you to our [cartesian_controllers](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers). That package brings interactive Cartesian control to the ROS-control framework. You can also use it for teleoperation on robots that use ROS-control (this driver included).
You can check [this short video](https://vimeo.com/378682968) from ROSCon last year to see if this covers your use case.

>  In particular, I am asking if I have to use a specific frequency to publish goal positions on the followings (and the reason for that):

With the [position_controllers/CartesianMotionController](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master/cartesian_motion_controller) you can decide yourself how frequent you want to publish target poses. This can be anything from sparsely sampled e.g. 1Hz to high-frequent 1000Hz. The controller offers an interpolation mechanism that makes sure that the driver's control cycle is happy at all times.

ManueU, 2020-11-19:
Thank you for your answer @stefanscherzinger!
I have some question for you.

1. If I want to teleoperate using the position_controllers/CartesianMotionController, I have to publish cartesian pose data to /my_cartesian_motion_controller/target_frame topic. Is it correct?
2. I see that types and names of the fields in the message are:

`std_msgs/Header header
  uint32 seq
  time stamp
  string frame_id
geometry_msgs/Pose pose
  geometry_msgs/Point position
    float64 x
    float64 y
    float64 z
  geometry_msgs/Quaternion orientation
    float64 x
    float64 y
    float64 z
    float64 w`

- I don't know what I have to put inside the header field.
- Is the position's unit of measurement in mm?
- Does the pose must be express in the reference system of the base of our robot? Does you know which is that in the case of the UR?

stefanscherzinger, 2020-11-19:
> If I want to teleoperate using the position_controllers/CartesianMotionController, I have to publish cartesian pose data to /my_cartesian_motion_controller/target_frame topic. Is it correct?

Yes, that's correct. You can also remap this topic if you wish.

> I don't know what I have to put inside the header field.

The only required information in the header (and the only one used) is _header.frame_id_. That's a sanity check to see if the incoming targets are given in the specified _robot_base_link_ (from the controllers.yaml). For UR robots a suitable choice could be _base_link_, if I'm not mistaken.

> Is the position's unit of measurement in mm?

The implementation sticks to the SI convention as proposed by ROS. So yes, It's meters for translation.

> Does the pose must be express in the reference system of the base of our robot? Does you know which is that in the case of the UR?

Yes, that's a requirement to accept the incoming target pose. The controller actually warns you if your according .yaml specification does not match the header's frame_id. For UR robots, I personally prefer _base_link_.

You could additionally visualize this target pose in Rviz to get a feeling of how that moves in the vicinity of your robot end-effector. If you fear that you could send infeasible poses, try keeping the controller's error_scale (dynamic_reconfigure) low, e.g. 0.05 in first tests. The robot will move very slowly, giving you sufficient time to react.

ManueU, 2020-11-23:
Thank you @stefanscherzinger for your fast reply.

I defined the cartesian_motion_controller in the `ur5_controllers.yaml` file. 
To start the robot driver I used one of the existing launch files: 
`$ roslaunch ur_robot_driver <robot_type>_bringup.launch robot_ip:=192.168.56.101 \
  kinematics_config:=$(rospack find ur_calibration)/etc/ur10_example_calibration.yaml`
I chosed the cartesian_motion_controller using the controller_manager. 

The robot moves, but it doesn't follow the distance that I sent in terms of cm. I read `examples.launch` file in the folder cartesian_controller_examples. Could it be a problem that depends on something that I haven't run? Do you have any suggestions to solve my problem?

ManueU, 2020-11-26:
Hi @stefanscherzinger, the previously mentioned problem about the strange movement of the robot is solved by setting the header's frame_id to "base". 
I have another question. You said to me: 

> With the position_controllers/CartesianMotionController you can decide yourself how frequently you want to publish target poses. This can be anything from sparsely sampled e.g. 1Hz to high-frequent 1000Hz. The controller offers an interpolation mechanism that makes sure that the driver's control cycle is happy at all times.

Could you give more details about the interpolation of data in your cartesian_motion_controller? 
For example: 
- What happens when the data are sent with a rate of 1000 Hz? 
- What happens when the data are sent with a low rate, for example, 1 Hz? 
Thank you for your time.

stefanscherzinger, 2021-08-09:
I'm closing this here since Cartesian controller internals are out of the focus of this repo. Feel free to open issues at the [cartesian_controllers](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers) repo instead. Maybe[ this issue](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/19) answers some of your questions.

closed 2021-08-09 by stefanscherzinger