UniversalRobots/Universal_Robots_ROS_Driver #290: Read return value from Module internals functions

ManciuCen, 2020-11-18:
Hi!

I have successfully installed  the Universal_Robots_ROS_Driver and I am able to control the robot via the action client or moveit package, but I was confused about how to read the return value from Module internals function like` force()` or `get_inverse_kin()`. When I use the command below:

- `rostopic pub /ur_hardware_interface/script_mmand std_msgs/String "data: 'get_inverse_kin(p[.1,.2,.2,0,3.14,0],[0.,3.14,1.57,.785,0,0])'"` 

It seems it got nothing return from any topic or service, I had been read about the question like #162 , and know that `movel()` or `movej()` receive nothing feedback, is this because the sending script from topic method does not support reading the return value or just because the `move*()` functions had no return value?

Best regards and thanks in advance for your help!

gavanderhoorn, 2020-11-18:
Disregarding the URScript and UR controller side of things: you are publishing to a topic. There is no way to receive any feedback, as it's a fire-and-forget interface.

ManciuCen, 2020-11-18:
Okay, got it. Thanks @gavanderhoorn.
Back to the question, is there any method that I can used in this driver to send URScript snippets and receive the return value easily (if it has return value)

fmauch, 2020-11-18:
Script code sent to the robot gets executed there. Results from script expressions only live inside the script context running on the robot. URScript is no remote call library.

If you want to receive data generated in URScript, you'll also have to use URScript to send it back from the script context. For example, you could connect to a socket and use one of the `socket_send...` functions.

ManciuCen, 2020-11-21:
Thanks for your reply! @fmauch  I have no questions! :)

closed 2020-11-21 by ManciuCen