UniversalRobots/Universal_Robots_ROS_Driver #291: ROS / ROS 2 Compatibility

vanthome, 2020-11-21:
Sorry in case I have missed it, but I cannot find witch which ROS Versions this module is compatible with. Can you please clarify?
BG

gavanderhoorn, 2020-11-21:
From [the readme](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/README.md#requirements):

> ### Requirements
> 
> This driver requires a system setup with ROS. It is recommended to use **Ubuntu 18.04 with ROS melodic**, however using Ubuntu 16.04 with ROS kinetic should also work.

gavanderhoorn, 2020-11-21:
Work for ROS 2 is underway, but not in a usable state right now.

gavanderhoorn, 2020-11-21:
And ee #69, which was even pinned.

vanthome, 2020-11-21:
OMG, sorry, I did not look at the pinned ones!

gavanderhoorn, 2020-11-21:
If your question has been answered, I'd suggest you close your issue.

closed 2020-11-21 by vanthome