UniversalRobots/Universal_Robots_ROS_Driver #293: Is there any method to set ft-sensor-wrench topic?

Csp0920, 2020-11-25:
I find that the ft sensor data is published to /wrench, and I want to change it, such as publishing ft sensor data to my_ft_sensor/wrench.

If I should modify the config in ur_robot_driver/config/ur3e_controllers.yaml?
force_torque_sensor_controller:
   type:         force_torque_sensor_controller/ForceTorqueSensorController  ------------------(modify this?)
   publish_rate: *loop_hz

gavanderhoorn, 2020-11-25:
You don't ever (well, perhaps maybe in some special case) have to change code to get ROS nodes to publish on some other topic.

Doing that is a very strong anti-pattern in ROS.

Use [remapping](http://wiki.ros.org/Remapping%20Arguments). See also [#q303611](https://answers.ros.org/question/303611) on ROS Answers.

Csp0920, 2020-11-25:
> You don't ever (well, perhaps maybe in some special case) have to change code to get ROS nodes to publish on some other topic.
> 
> Doing that is a very strong anti-pattern in ROS.
> 
> Similarly, don't ever add configuration options (ie: parameters) to nodes to set names of topics.
> 
> So the `ForceTorqueSensorController` does not support doing what you ask.
> 
> Use [remapping](http://wiki.ros.org/Remapping%20Arguments). See also [#q303611](https://answers.ros.org/question/303611) on ROS Answers.
> 
> Edit: or alternatively, use namespacing.

Can remap be added to any node? Or can remap only in the node where this topic is published？

Csp0920, 2020-11-25:
Original launch:
<node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="$(arg controllers)" />

Modified launch:
<node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="$(arg controllers)" >
    <remap from="/wrench" to="/my_ft_sensor/wrench" />
</node>

Is that what you said？

gavanderhoorn, 2020-11-25:
You need to remap the topic for the node which is actually publishing to it.

The `controller_manager` does not publish to `wrench`. That would be the driver itself.

Csp0920, 2020-11-25:
> 您需要重新映射实际发布到该节点的主题。
> 
> 在`controller_manager`不发布到`wrench`。那将是驱动程序本身。

 <arg name="controllers" default="joint_state_controller scaled_pos_joint_traj_controller speed_scaling_state_controller force_torque_sensor_controller robot_status_controller" doc="Controllers that are activated by default."/>

I find that controller_manager spawns force_torque_sensor_controller. So I think the topic is published here.
Is it right?

gavanderhoorn, 2020-11-25:
No.

"spawning" just means instructing the `ControllerManager` class *inside* the driver to load a `.so` and instantiating an object from that `.so`.

The code "runs inside" the driver node.

Csp0920, 2020-11-25:
> No.
> 
> "spawning" just means instructing the `ControllerManager` class _inside_ the driver to load a `.so` and instantiating an object from that `.so`.
> 
> The code "runs inside" the driver node.

But I didn't really see where /wrench was published...

Csp0920, 2020-11-25:
Is it in here?
<node name="ur_hardware_interface" pkg="ur_robot_driver" type="ur_robot_driver_node" output="screen" launch-prefix="$(arg launch_prefix)" required="true">

gavanderhoorn, 2020-11-25:
Yes, that's the node which hosts the `ControllerManager` and loads the F/T controller plugin.

Csp0920, 2020-11-25:
> 是的，那是托管`ControllerManager`并加载F / T控制器插件的节点。

Thank you very much for your patient.

Csp0920, 2020-11-30:
Hello, after I remap “/wrench” to "/new_wrench".
I got the error: "pipeline producer overflowed! <RTDE Data Pipeline>"
And I felt that my terminal became very laggy.
I see the question＃6. He solved the problem by replacing the PC.
If  this problem related to PC？

gavanderhoorn, 2020-11-30:
I have not encountered this, so I would not know what the cause would be.

Perhaps @fmauch knows something.

Remapping by itself should not cause this, as it's a one-time thing.

fmauch, 2020-11-30:
No, I don't. But I'll look into it.

fmauch, 2020-11-30:
> Hello, after I remap “/wrench” to "/new_wrench".
> When I tried to `rostopic echo /new_wrench`
> I got the error: `pipeline producer overflowed! <RTDE Data Pipeline>`
> And I felt that my terminal became very laggy.
> I see the question＃6. He solved the problem by replacing the PC.
> If this problem related to PC？

Just a couple of questions:
 - Does the `rostopic echo` work without the remap? I would suggest, it behaves the same, as @gavanderhoorn mentioned.
 - Does this happen every time?
 - Do you use a real-time kernel as [suggested](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#requirements)?
 - What is the system load on your system? (You could post the output of `top -n 1`)

Csp0920, 2020-12-01:
1. Does the `rostopic echo` work without the remap?
I deleted the remap code and tried it, and it won't report a error `pipeline producer overflowed! <RTDE Data Pipeline>`.
However, it is disconnected from the PC from time to time. `Connection to robot dropped, waiting for new connection.`
2. Does this happen every time?
I've tried both cases many times. They happened every time.
3. Do you use a real-time kernel as suggested?
I think I have used rt knernel.
[image]
4. What is the system load on your system?
[image]

Csp0920, 2020-12-01:
[image]
[image]

Csp0920, 2020-12-28:
Hello, after using another better computer, the error will not be happened again.
Thanks for your help.

closed 2021-08-09 by stefanscherzinger