UniversalRobots/Universal_Robots_ROS_Driver #294: Strange FT sensor readings with nothing mounted on the end of the arm

carebare47, 2020-11-25:
# Summary

We're trying to use the FT sensor for compliance control. At the moment there is nothing mounted on the end of the arm. 

If we start the arm with the end link pointing outward, zero the FT sensor, and then rotate wrist 3 by 90 degrees, we get a reading of around 10N on `Y`. 

But, if we start with the end link pointing upwards, and rotate wrist 3, we see something similar (X: -2, Y:8, Z:1.2). This position should negate the effects of gravity on the FT sensor, so why are the readings changing? Is this expected behaviour? 

We're using a UR10e and reading FT values from the `wrench` topic (have checked the values coming from the RTDE packet, and aside from the conversion in the `transformForceTorque()` function, the values we get from the packet are the same as those on the `wrench` topic).

fmauch, 2020-11-25:
As you have already stated, we only apply the coordinate system transformation to the data reveived from the RTDE interface. On the data sent to the RTDE interface this driver doesn't have any influence.

carebare47, 2020-11-25:
So the driver isn't having any influence on these values. Have you seen this kind of behaviour before? If not I'll direct my question to UR support and post my findings back here

fmauch, 2020-11-25:
Did you setup your payload (mass and cog) correctly on the robot? In your case they should all be zero if I understand the OP correctly.

Apart from this I don't have an idea where that is coming from.

carebare47, 2020-11-25:
yeah, mass and cog are all zero and there's nothing physically attached to the eef. I'm talking with UR, they're saying this sounds weird. I'll post here whether it turns out to be a problem with our arm or a general issue

fabioalves98, 2020-12-09:
I am having the exact same problem. Also using a UR10e, also trying to read the values from the FT sensor and everytime we apply a rotation to the EE, the values on each axis change in a very strange pattern.

[image]
(X - Red, Y - Green, Z - Blue) Wrist 3 starts with a joint value of 0

This is an example of such pattern. Even if I record this values with a different orientation of the EE, different payload or cog, the pattern repeats itself (with silight amplitude differences)
I know it is not the driver's fault, since it only forwards the values from the UR10e internal controller and transforms them.
For now, to fix the issue, I am calling the zero_ftsensor service to emilinate this discrepancies but it's really not ideal.
Have you heard an news from UR?

stefanscherzinger, 2021-08-09:
@carebare47 Any news from your side?

carebare47, 2021-08-09:
Hi all, sorry for not updating you all sooner. Confirmed issue from UR - we are now using an external sensor