UniversalRobots/Universal_Robots_ROS_Driver #296: Using this ROS Driver without externalcontrol-1.0.4.urcap

ManciuCen, 2020-12-01:
Hi!
When I using this driver with externalcontrol-1.0.4.urcap, I don't want to use ros_control framework to control the robot. So I write a *.urp (basically include socket functions and move functions) program on the polyscope to respond the request from the remote control pc. But when I put the externalcontrol-1.0.4.urcap into the *.urp program and start running, a popup dialog shows **"Another thread is already controlling the robot"**, my question is:
1. Is that means I can't add any other move* functions on the program that included externalcontrol-1.0.4.urcap?
2. What features will be lost when I try to use Universal_Robots_ROS_Driver without externalcontrol-1.0.4.urcap? It seems I can't get any feedback from /speed_scaling_factor anymore, it is always zero.
3. How can I get the RTDE data like (actual_TCP_pose, speed_scaling, target_speed_fraction) with this ROS driver when using my own *.urp program on the polyscope(Is there any other functions that I need to add?)

Best regards and thanks in advance for your help! :P

ManciuCen, 2020-12-01:
> It seems I can't get any feedback from /speed_scaling_factor anymore, it is always zero

update the question-2, /speed scaling_factor topic will be updated as long as I play my *.urp program on the polyscope.

gavanderhoorn, 2020-12-01:
Afaik (but we'll let @fmauch confirm): you'll lose the ability to *control* the robot. All state reporting should still work.

> `/speed scaling_factor` topic will be updated as long as I play my *.urp program on the polyscope.

this does make sense in a way, as there is no speed scaling active when the robot is stationary.

fmauch, 2020-12-01:
> * Is that means I can't add any other move* functions on the program that included externalcontrol-1.0.4.urcap?

The externalcontrol urcap is only for controlling the robot from an external site, currently only motion streaming is supported. If you don't want to use this, don't include it. You can still use the driver to read data from the robot, but you won't be able to control it using ROS without doing manual work.

> * What features will be lost when I try to use Universal_Robots_ROS_Driver without externalcontrol-1.0.4.urcap? It seems I can't get any feedback from /speed_scaling_factor anymore, it is always zero.

As @gavanderhoorn already said, you lose the ability to control the robot from ROS. `/speed_scaling_factor` isn't connected with this. It's just how the speed scaling logic on UR robots work. No motion command, no speed scaling. You will also notice that when a program is paused or a protective stop / EM-stop is triggered, it will become 0.

> * How can I get the RTDE data like (actual_TCP_pose, speed_scaling, target_speed_fraction) with this ROS driver when using my own *.urp program on the polyscope(Is there any other functions that I need to add?)

RTDE data streaming is absolutely possible without using the externalcontrol urcap. As I said earlier, you "only" lose the functionality to control the robot using this driver. The driver will still publish data received from the RTDE interface.

ManciuCen, 2020-12-05:
Thank you so much! @fmauch @gavanderhoorn

closed 2020-12-05 by ManciuCen