UniversalRobots/Universal_Robots_ROS_Driver #297: error in installing Universal_Robots_ROS_Driver

abdul-mannan-khan, 2020-12-01:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*

# Versions
 - ROS Universal_Robots_ROS_Driver:
 - Affected Robot Software Version(s):
 - Affected Robot Hardware Version(s):
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):

# Impact
*What is the impact of the issue. Try to describe it well enough, in order for us to prioritise the issues.*

# Issue details
*This part is important in order to clarify the severity of the issue.*

## Use Case and Setup
*What does the setup look like and what are the objectives?*

## Project status at point of discovered
*When did you first observe the issue?*
 - *Before I started?*
 - *In first couple of tries?*
 - *In normal use?*
 - *etc...*

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

fmauch, 2020-12-01:
This looks like you didn't clone the `universal_robots` repository from my fork as explained in the [installation guide](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building), particularly this section:

```bash
# clone fork of the description. This is currently necessary, until the changes are merged upstream.
$ git clone -b calibration_devel https://github.com/fmauch/universal_robot.git src/fmauch_universal_robot
```

closed 2020-12-02 by abdul-mannan-khan

gavanderhoorn, 2020-12-02:
And just making sure:

> I followed [this](http://wiki.ros.org/universal_robot/Tutorials/Getting%20Started%20with%20a%20Universal%20Robot%20and%20ROS-Industrial) tutorial. I am using UR 5 and I want to interface it with my ROS-melodic. Following the tutorial, I cam across the instruction to install [ur_modern_driver](https://github.com/ros-industrial/ur_modern_driver). However, upon checking the link, I found that these drivers are deprecated.

Did you notice the deprecation warning on the tutorial you link?

[image]

The tutorial itself already mentions `ur_modern_driver` must not be used and also deprecates the tutorial itself.